openscad/openscad #4452: 2021.01.01 fails to build with CGAL-5.5.1: no template named 'Triangulation_2_filtered_projection_traits_3' in namespace 'CGAL'; did you mean 'Triangulation_2_projection_traits_3'?

yurivict, 2022-12-19:
```
src/cgalutils-tess.cc:22:15: error: no template named 'Triangulation_2_filtered_projection_traits_3' in namespace 'CGAL'; did you mean 'Triangulation_2_projection_traits_3'?
typedef CGAL::Triangulation_2_filtered_projection_traits_3<K> Projection;
        ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
              Triangulation_2_projection_traits_3
/usr/local/include/CGAL/Triangulation_2_projection_traits_3.h:32:7: note: 'Triangulation_2_projection_traits_3' declared here
class Triangulation_2_projection_traits_3
      ^
```

clang-14
OS: FreeBSD 13.1

t-paul, 2022-12-21:
Not much we can do here, the latest code has fixes for that.

A list of patches backported for the Debian builds of release 2021.01 can be found at https://sources.debian.org/patches/openscad/2021.01-5/

yurivict, 2022-12-21:
You should make a new release then. Last release is rendered broken by these changes in CGAL.

closed 2022-12-21 by t-paul