openscad/openscad #4455: Methods

jordanbrown0, 2022-12-20:
Once we have objects (or dictionaries if you prefer), they can hold function references, and so you can say things like `obj.method()`.  A simple extension of that capability would have it define a `$self` or `$this` variable that refers to the object that the function reference came from.

But it would also be good to be able to use the same syntax on other kinds of data values, so that we could add new operations without consuming global namespace.  It would be nice to be able to add a host of string operations, for instance, without needing to consume global namespace for each of them.  One could imagine `[...].sum()`, for instance, or `number.cosh()`.

For extra credit, figure out a way for a program to add its own methods to existing types.

kwikius, 2022-12-20:
Use of $this or $self, to access the object, the function is called on, sounds good to me.
You could presumably lookup any .fun() that is called, but would have to figure out what happens when both the object and the "builtin" are found. Presumably the user defined one wins to avoid surprises.  Anyway for "what could possibly go wrong? " ,  look at swizzle regex for arrays. Please don't go down that road :0.

jordanbrown0, 2022-12-29:
I put together an implementation of this feature, but in trying to come up with a realistic example I found that most of the time it's not necessary.

Most of the time, the object is created and the function is declared in the same scope.  The function declaration captures that scope, which means that all of the information that went into the object is also available to the function.

Only if the function is declared in a different scope would it be irreplaceable to have access to the object.

jordanbrown0, 2022-12-30:
One use case (sort of) comes to mind, similar to the "prototype" concept in JavaScript:

```
proto = { method1: function (...) ..., method2: function (...) ... };
...
obj = object(proto, val1=..., val2=...);
```

Where proto in a sense defines a class, and you use object() to add instance-specific values.

However, you probably want to wrap the object() call in a factory/constructor function, at which point it might as well populate the methods at the same time without having a prototype object.