openscad/openscad #4458: New file boilerplate

jon-bondy, 2022-12-23:
I start all of my OpenSCAD programs the same way, establishing some constants (like "inches = 25.4;") and importing/using some files.  I type this in by hand, over and over and over again.

It would be nice if, when OS created a New file, and if a file like "Default.scad" is present in (say) the Library folder, then the New file is filled with the contents of the Default.scad file.  Since this file will not exist on older installations, this new feature will only affect those who wish to use it going forward.

Thanks for considering this.

Jon

jordanbrown0, 2022-12-23:
Might be worthwhile, but note that you can reduce your per-file manual work to
```
include <Default.scad>
```
Note that then if you want to make a change to your boilerplate, it immediately affects all of your models.  That might be good, or might be bad.

t-paul, 2022-12-24:
You can store your own code snippets in the config folder, e.g. on Linux `.config/OpenSCAD/templates` and insert those via `ALT`+`INS` or Context-Menu->Insert Template.

Example files: https://github.com/openscad/openscad/tree/master/templates

jon-bondy, 2022-12-24:
Nice.  I had no idea this was possible.  But I like my idea better.  Of 
course I do!

Thanks!

t-paul, 2022-12-24:
Yes, I think we can keep this open as separate feature. However I would not go with a single fixed file, but instead something like "New from template" which can select from files either shipped or user provided.

bitbasher, 2025-10-15:
/assign-me

jon-bondy, 2025-10-15:
I suppose we could add a new field to the JSON record called "tag" which could be blank (meaning put this in the list when the user attempts an Insert Template), or "new" (meaning put this in the list when New From Template is used) or "default" (meaning insert this every time a new file is created.  Then the user is in complete control, assuming they are able to locate and edit the files.

Jon

On 10/15/2025 5:23 PM, Jeff Hayes wrote:
[https://avatars.githubusercontent.com/u/430590?s=20&v=4]bitbasher left a comment (openscad/openscad#4458)<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_openscad_openscad_issues_4458-23issuecomment-2D3408350640&d=DwMCaQ&c=euGZstcaTDllvimEN8b7jXrwqOf-v5A_CdpgnVfiiMM&r=AsrE-c7ZR7B2Kyr3qgfvvppkCEBVsNmwEMndcrRSuOI&m=oI-q6xkUF0NDQJsgyhqUr0VfmPV20xzw_eSoiE0BgvQacM_6LYXCDNhkKTicqnI7&s=MrjgJXpo58fnHlEpol-8_u5QiKSlsjP--zNfkPCcJtU&e=>

i think we can do both

Part 1
the user can create a specially named file like /User/vulcan_/.config/OpenSCAD/templates/UserDefinedStartTemplate.json
It will be easy to notice this special file as the templates are being loaded by Scintilla as it starts up .. it could even have a preference checkbox to enable the auto-loading template ( .. and a text field to specify the filename? possible but is it too much? )

should that be Enabled by Default ? or require the user to enable it explicitly?

Part 2
a new menu File > New From Template .. pop-up, select from list, and paste it into place once the editor tab is open

this is not going to be much use if templates are simple strings that need explicit \n \t for formatting .. i think i would want to add the possibility of multi-line text templates. If that is the case .. would i want to allow include <> and use <> to operate when pasted into a new editor tab .. or should they be excluded from .scad text blocks?

this would also mean having a section on the start up splash window for templates .. either widen the splash to have space for three selection lists .. or just have one area for selections with three tabs?