openscad/openscad #4459: Infinite recursion with too many function arguments hangs

jordanbrown0, 2022-12-23:
This program gives five  "too many unnamed arguments supplied" errors and then goes unresponsive:
```
function foo()  = foo(0);
echo(foo());
```
But these two programs do not; they get a "recursion detected" error:
```
function foo(x)  = foo(0);
echo(foo());
```
```
function foo()  = foo();
echo(foo());
```

jordanbrown0, 2022-12-23:
Nor does this one; it gets one "Too many unnamed arguments" error, pauses, and then gets "recursion detected":
```
function foo(a)  = foo(0,1);
echo(foo());
```

This program aborts after a million recursions.  On my system, it takes 7.5 seconds.
```
$i = 0;
function foo() = let(dummy=$i%1000==0?echo($i):0) let($i=$i+1) foo();
echo(foo());
```

On the other hand, this program, with the argument error, only seems to process about 333 recursions per second; it prints a number every ~3 seconds.
```
$i = 0;
function foo() = let(dummy=$i%1000==0?echo($i):0) let($i=$i+1) foo(0);
echo(foo());
```

So probably the original program isn't hung forever... it just takes enormously longer to fail than the other cases do.

thehans, 2022-12-24:
This is entirely due to the "Error Log", which seems to take linear time to append each error.  This is why we should probably have some configurable "Max Error Limit" in preferences (with a default of maybe 100).
This could also potentially replace the "stop on first warning" option (by setting the value to 1).

thehans, 2022-12-24:
I wrote about this when I was working on increasing the GUI Console performance.  
See the second half of my comment here: https://github.com/openscad/openscad/pull/3602#issuecomment-761282448

gsohler, 2024-08-07:
Did you consider to Put the Warnung creating Code into a Thread so IT will Not alle down the GUI ?

thehans, 2024-08-07:
The slowness is in the GUI code itself (updating the Error Log's QGridLayout widget) AFAIK.  Qt GUI code all happens on a single main thread so it can't be offloaded.

gsohler, 2024-08-07:
Ahh, now I understand,
The big time delay happens, when a big amount of warning issues are
consumed in the GUI by re-rendering the GridLayout,
That makes perfect sense.

Still putting the error-generating code into a separate thread would make
sense.

OpenSCAD would be about 10 times faster (example given)
if it would process a chunk of 10 warning messages at once(and
recalculating Gridlayout once)
instead of
updating the qgrid separately for each new warning messages.
There could be a QTimner to update the Log field periodically(at most every
0.5 sec e.g.) and it would be fast enough

just my 2 little cents