openscad/openscad #4460: Nested modules are not allowed inside if() or union()

odisseus, 2022-12-25:
# The problem

This code compiles:
```
module foo() {
    module bar() {}
}
```

But this produces an error:
```
module foo() {
    if(1 == 1) {
        module bar() {}
    }
}
```

This also produces an error:
```
module foo() {
    union() {
        module bar() {}
    }
}
```

# Justification
When writing some complicated code which has a lot of condition checking and nested unions and differences, I often want to extract repeating parts into a module. I would like to keep that module close to the place where it's actually used, which is often a single place that's nested in a few levels of `if()` and `union()`. This being impossible, I have to place the module outside the topmost `if()`, which makes the code harder to read and exposes irrelevant details to the top-level modules.

UBaer21, 2022-12-25:
If you create a module it is there - this is not conditional. It is a functional language so it can not change depending on the state of your program. This is also why all variables are constants. 
You can use the module conditional but not create it. Also creating modules in a loop would result in multiple modules with the same name. 
You can create a module within a module so it is within a different scope. 
You can not have a union with the creation of a module as the creation of a module is not an object - only the module created is.
You also can not make a variable (constant) conditional with if only the scope it is in.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_Functions_and_Modules

kwikius, 2022-12-25:
You can do the following on my fork, with  so called module_literal. I hope the code gives you an indication of how it works. You can do many things with module_literal/module_reference including putting modules in arrays and in if else statements.

https://github.com/openscad/openscad/pull/4367
https://github.com/kwikius/openscad/tree/module_literal_v3

```openscad
module foo(b ) {

    if(b == true) {
        bar = module cube([10,20,30]);
        bar();
    }
    
   
    union() {
        bar = module () {
          translate([0,0,10]) sphere(r = 5);
        };
        bar();
    }
}

foo(true);
```
In fact you can even declare modules inside  function arguments. The following shows an alternative to difference using the code on the below fork
```

   module subtract(a,b){
     difference(){
       a();
       b();
     }
   };

   subtract(
      module { 
         translate([-20,0,-20])
            cube([40,40,40]);
      }, 
      module sphere(d=20)
   );
```

kwikius, 2022-12-25:
Incidentally a shorthand without requiring to declare a variable is as follows. The syntax (putting the module expression in parentheses) is somewhat cryptic to avoid large modification to the openscad parser, but it gets the job done!
```openscad
module foo(b ) {

    if(b == true) {
        ( module cube([10,20,30]) )();
    }
```

closed 2026-05-29 by github-actions[bot]