openscad/openscad #4479: error compiling on macOS

dwightkelly, 2022-12-29:
Apple clang version 14.0.0 (clang-1400.0.29.202)
Target: x86_64-apple-darwin21.6.0
OS: macOS 12.6.1 Intel
openscad @2015.03-3_9 (installed using macport)

```
:info:build In file included from src/LibraryInfo.cc:15:
:info:build In file included from src/cgal.h:24:
:info:build In file included from /opt/local/include/CGAL/Gmpq.h:26:
:info:build In file included from /opt/local/include/CGAL/Gmpz.h:34:
:info:build In file included from /opt/local/include/CGAL/Gmp_coercion_traits.h:33:
:info:build /opt/local/include/CGAL/GMP/Gmpfr_type.h:208:25: error: cannot initialize a variable of type 'mpfr_ptr' (aka '__mpfr_struct *') with an lvalue of type 'mpfr_srcptr' (aka 'const __mpfr_struct *')
:info:build                         mpfr_custom_get_kind(f),
:info:build                         ^~~~~~~~~~~~~~~~~~~~~~~~
:info:build /opt/local/include/mpfr.h:1030:14: note: expanded from macro 'mpfr_custom_get_kind'
:info:build     mpfr_ptr _x = (x);                                                  \
:info:build              ^
:info:build /opt/local/include/mpfr.h:1010:9: note: expanded from macro 'mpfr_custom_init_set'
:info:build   _k = (k);                                                    \
:info:build         ^
```

kintel, 2022-12-29:
"openscad@2015.03-3_9" - this is an ancient version of OpenSCAD. If you really want to get this working on MacPorts, I would suggest either upgrading the MacPorts build or work with the MacPorts maintainer (if they're still active), or report a bug against MacPorts: https://ports.macports.org/port/openscad/

For an easier path, we do offer pre-compiled binaries at https://openscad.org/downloads.html

dwightkelly, 2022-12-29:
I would recommend deleting the macport listing on your website.

[image]

t-paul, 2022-12-31:
CGAL issue mentions this is actually a MPFR-4.1.1 bug - https://github.com/CGAL/cgal/issues/7064

kintel, 2022-12-31:
Good find @t-paul - I've verified that building against mpfr-4.1.1 and cgal-5.5 fails on master as well.
Reopening and renaming for tracking.

kintel, 2022-12-31:
..but since neither MacPorts or Homebrew packages are maintained, we should perhaps not recommend them.

PR #4496 (2023-01-13): Bump mpfr to 4.2.0
Fixes #4479
closed 2023-01-13 by kintel