openscad/openscad #4484: setenv-unibuild.sh fails when PATH contains a space

durquhart, 2023-01-01:
scripts/setenv-unibuild.sh fails when run from scripts/uni-build-dependencies.sh:737 if PATH contains a space.

While a space in the path is a bad idea in general, it was put there by the NVIDIA CUDA driver install on my system, which means I'm unlikely to be the only person to run into this.

There's a simple 1-line fix. Diff below. Interestingly this doesn't seem to have caused any further build problems.

I'm happy to put this in a PR as well, not sure what you prefer.
I know this is a simple newbie problem, but smooth setup makes it easier for the next person to come onboard and get to more interesting contributions.

Observed on 
 master 318aac9cc 

----

```
diff --git a/scripts/setenv-unibuild.sh b/scripts/setenv-unibuild.sh
index 0b3fcd79b..29ad3886b 100644
--- a/scripts/setenv-unibuild.sh
+++ b/scripts/setenv-unibuild.sh
@@ -20,7 +20,7 @@ setenv_common()
  DEPLOYDIR=$BASEDIR

  export BASEDIR
- export PATH=$BASEDIR/bin:$PATH
+ export PATH=$BASEDIR/bin:"$PATH"
  export LD_LIBRARY_PATH=$DEPLOYDIR/lib:$DEPLOYDIR/lib64
  export LD_RUN_PATH=$DEPLOYDIR/lib:$DEPLOYDIR/lib64
  export OPENSCAD_LIBRARIES=$DEPLOYDIR
```

kintel, 2023-01-01:
This is kind of surprising, as quoting variable expansions in assignments shouldn't be necessary. Are you using an unusual shell perhaps, which doesn't implement Posix behavior?

Otoh, I don't see any immediate danger in adding extra quotes. Opening a PR for visibility would be great.

durquhart, 2023-01-04:
Put up the 1-line PR: https://github.com/openscad/openscad/pull/4489

I'm running Ubuntu under WLS2, which might be a bit niche.  The shell is
`GNU bash, version 5.0.16(1)-release (x86_64-pc-linux-gnu)`

closed 2023-01-04 by kintel