openscad/openscad #4485: customizer slider slips into only evens when stepping

Neon22, 2023-01-01:
I wanted to have a slider which started at 15, stepped up in twos to 199. So only showed odd numbers.
However this:
```Thread_count = 49; // [15:2:199]```
slips into even numbers when using it.

thehans, 2023-01-03:
Please try the most recent stable release: 2021.01
If that still has problems, then include more information about your system.  Open "Help" -> "Library Info" and paste the contents here.

I am not able to reproduce on Linux Mint 21 using OpenSCAD 2021.01, nor on nightly snapshots.

Neon22, 2023-01-03:
sorry - yes working properly in 2021.01 on Windows.
not sure what page I was on but went to homepage and saw 2019 was latest which agreed with my aboutbox.
but now I see 2021.01 is latest. Sorry I didn't record the url I visited :(
Closing

Neon22, 2023-01-04:
reopening.
The trick to revealing this bug is to enter an even number manually into the slider.
Then the slider steps by 2 but now is even. Does not force back to odd.

Its not clear this is entirely wrong... some determination could be made.
But it breaks my specific example badly. I'd vote for manual entry to be overridden to follow the author's intention.