openscad/openscad #4486: Numerous test failures when enabling `CGAL_CHECK_EXPENSIVE` compile defintion

thehans, 2023-01-02:
I was reading CGAL documentation regarding [Checks: Pre- and Postconditions, Assertions, and Warnings](https://doc.cgal.org/latest/Manual/devman_checks.html#secchecks_using) which explains that there are four types of checks:
 - `CGAL_<check_type>`
 - `CGAL_expensive_<check_type>`
 - `CGAL_exactness_<check_type>`
 - `CGAL_expensive_exactness_<check_type>`

> By default, all standard checks (without any attribute) are enabled, while expensive and exactness checks are disabled.
> ...
> To enable expensive and exactness checks, respectively, the compile-time flags `CGAL_CHECK_EXPENSIVE` and `CGAL_CHECK_EXACTNESS` have to be supplied.

So, realizing that there are additional checks which might help us track down the source of problems earlier on, I gave it a try and found that `CGAL_CHECK_EXPENSIVE` caused 111 of our default test cases to fail.

A summary of assertion failures follows (some tests may have multiple occurrences of the same assertion failure, or even multiple assertion failures of different types).

105 occurrences (in 84 tests) of:
```
CGAL ERROR: assertion violation!
Expr: FPU_get_cw() == CGAL_FE_TONEAREST
File: /usr/include/CGAL/Filtered_predicate.h
Line: 110
```

35 occurrences of (in 19 tests) of:
```
CGAL ERROR: postcondition violation!
Expr: is_valid()
File: /usr/include/CGAL/Nef_polyhedron_3.h
Line: 1108
```

13 occurrences (in 6 tests) of:
```
CGAL ERROR: assertion violation!
Expr: FPU_get_cw() == CGAL_FE_PROTECTED
File: /usr/include/CGAL/FPU.h
Line: 568
```

So it seems that most of the issues are due to `FPU_CW_t FPU_get_cw (void)` returning an unexpected value.
The [documentation for this function](https://doc.cgal.org/latest/Number_types/classCGAL_1_1Interval__nt.html#addd87a741183ee6db8cbf70fa95e1a72) says:
> returns the current rounding mode.

It is unclear to me if this is some fault of OpenSCAD code, or some bugs in CGAL, but I'm not aware of any OpenSCAD sources touching the rounding mode at all.  So at first glance it seems like CGAL is at fault by not setting their own expected rounding mode somehow.
The only other possibility I can think of is that some other library such as Eigen or GMP/MPFR is possibly setting a rounding mode on its own, and not returning to the previous setting before returning control to CGAL.

List of failing tests:
```
        326 - cgalpngtest_projection-tests (Failed)
        379 - cgalpngtest_background-modifier (Failed)
        383 - cgalpngtest_child-modifier (Failed)
        391 - cgalpngtest_edge-cases (Failed)
        392 - cgalpngtest_for-tests (Failed)
        394 - cgalpngtest_highlight-modifier (Failed)
        396 - cgalpngtest_hull3-tests (Failed)
        402 - cgalpngtest_intersection_for-tests (Failed)
        405 - cgalpngtest_minkowski3-tests (Failed)
        414 - cgalpngtest_polyhedron-nonplanar-tests (Failed)
        416 - cgalpngtest_primitive-inf-tests (Failed)
        420 - cgalpngtest_rotate-parameters (Failed)
        431 - cgalpngtest_union-tests (Failed)
        441 - cgalpngtest_fn_bug (Failed)
        449 - cgalpngtest_issue1089b (Failed)
        451 - cgalpngtest_issue1105b (Failed)
        455 - cgalpngtest_issue1138 (Failed)
        465 - cgalpngtest_issue1225 (Failed)
        483 - cgalpngtest_issue911 (Failed)
        484 - cgalpngtest_issue913 (Failed)
        498 - cgalpngtest_localfiles-test (Failed)
        499 - cgalpngtest_localfiles-compatibility-test (Failed)
        532 - opencsgtest_projection-tests (Failed)
        603 - opencsgtest_hull3-tests (Failed)
        613 - opencsgtest_minkowski3-tests (Failed)
        663 - opencsgtest_issue1089b (Failed)
        669 - opencsgtest_issue1138 (Failed)
        697 - opencsgtest_issue911 (Failed)
        698 - opencsgtest_issue913 (Failed)
        741 - throwntogethertest_projection-tests (Failed)
        812 - throwntogethertest_hull3-tests (Failed)
        823 - throwntogethertest_minkowski3-tests (Failed)
        873 - throwntogethertest_issue1089b (Failed)
        879 - throwntogethertest_issue1138 (Failed)
        908 - throwntogethertest_issue911 (Failed)
        952 - csgpngtest_projection-tests (Failed)
        1005 - csgpngtest_background-modifier (Failed)
        1009 - csgpngtest_child-modifier (Failed)
        1017 - csgpngtest_edge-cases (Failed)
        1018 - csgpngtest_for-tests (Failed)
        1020 - csgpngtest_highlight-modifier (Failed)
        1022 - csgpngtest_hull3-tests (Failed)
        1028 - csgpngtest_intersection_for-tests (Failed)
        1031 - csgpngtest_minkowski3-tests (Failed)
        1040 - csgpngtest_polyhedron-nonplanar-tests (Failed)
        1047 - csgpngtest_rotate-parameters (Failed)
        1058 - csgpngtest_union-tests (Failed)
        1068 - csgpngtest_fn_bug (Failed)
        1076 - csgpngtest_issue1089b (Failed)
        1078 - csgpngtest_issue1105b (Failed)
        1082 - csgpngtest_issue1138 (Failed)
        1092 - csgpngtest_issue1225 (Failed)
        1110 - csgpngtest_issue911 (Failed)
        1111 - csgpngtest_issue913 (Failed)
        1125 - csgpngtest_localfiles-test (Failed)
        1126 - csgpngtest_localfiles-compatibility-test (Failed)
        1171 - lazyunion-opencsg_lazyunion-hull-for (Failed)
        1183 - lazyunion-cgalpng_lazyunion-hull-for (Failed)
        1185 - lazyunion-cgalpng_lazyunion-intersection-for (Failed)
        1186 - lazyunion-cgalpng_lazyunion-difference-for (Failed)
        1194 - lazyunion-monotonepng_lazyunion-hull-for (Failed)
        1196 - lazyunion-monotonepng_lazyunion-intersection-for (Failed)
        1197 - lazyunion-monotonepng_lazyunion-difference-for (Failed)
        1205 - lazyunion-stlpngtest_lazyunion-hull-for (Failed)
        1207 - lazyunion-stlpngtest_lazyunion-intersection-for (Failed)
        1208 - lazyunion-stlpngtest_lazyunion-difference-for (Failed)
        1216 - lazyunion-offpngtest_lazyunion-hull-for (Failed)
        1218 - lazyunion-offpngtest_lazyunion-intersection-for (Failed)
        1219 - lazyunion-offpngtest_lazyunion-difference-for (Failed)
        1225 - fastcsg-cgalpng_issue1455 (Failed)
        1227 - fastcsg-cgalpng_issue1138 (Failed)
        1229 - fastcsg-cgalpng_CSG-modules (Failed)
        1230 - fastcsg-cgalpng_CSG (Failed)
        1231 - fastcsg-cgalpng_logo (Failed)
        1232 - fastcsg-cgalpng_example001 (Failed)
        1235 - fastcsg-cgalpng_example004 (Failed)
        1236 - fastcsg-cgalpng_example005 (Failed)
        1237 - fastcsg-cgalpng_example006 (Failed)
        1238 - fastcsg-cgalpng_example012 (Failed)
        1243 - fastcsg-cgalpng_highlight-modifier (Failed)
        1244 - fastcsg-cgalpng_highlight-modifier2 (Failed)
        1247 - fastcsg-cgalpng_issue1105b (Failed)
        1254 - fastcsg-cgalpng_issue911 (Failed)
        1255 - fastcsg-cgalpng_issue913 (Failed)
        1256 - fastcsg-cgalpng_view-options-tests (Failed)
        1267 - fastcsg-lazyunion-cgalpng_lazyunion-hull-for (Failed)
        1269 - fastcsg-lazyunion-cgalpng_lazyunion-intersection-for (Failed)
        1270 - fastcsg-lazyunion-cgalpng_lazyunion-difference-for (Failed)
        1278 - fastcsg-lazyunion-amfpngtest_lazyunion-toplevel-objects (Failed)
        1282 - fastcsg-lazyunion-amfpngtest_lazyunion-hull-for (Failed)
        1284 - fastcsg-lazyunion-amfpngtest_lazyunion-intersection-for (Failed)
        1285 - fastcsg-lazyunion-amfpngtest_lazyunion-difference-for (Failed)
        1297 - fastcsg-lazyunion-3mfpngtest_lazyunion-hull-for (Failed)
        1299 - fastcsg-lazyunion-3mfpngtest_lazyunion-intersection-for (Failed)
        1300 - fastcsg-lazyunion-3mfpngtest_lazyunion-difference-for (Failed)
        1325 - monotonepngtest_issue1105b (Failed)
        1330 - monotonepngtest_polyhedron-nonplanar-tests (Failed)
        1345 - monotonepngtest_issue1225 (Failed)
        1347 - monotonepngtest_fn_bug (Failed)
        1354 - stlcgalpngtest_issue1105b (Failed)
        1359 - cgalstlcgalpngtest_polyhedron-nonplanar-tests (Failed)
        1374 - cgalstlcgalpngtest_issue1225 (Failed)
        1376 - cgalstlcgalpngtest_fn_bug (Failed)
        1377 - cgalbinstlcgalpngtest_polyhedron-nonplanar-tests (Failed)
        1392 - cgalbinstlcgalpngtest_issue1225 (Failed)
        1394 - cgalbinstlcgalpngtest_fn_bug (Failed)
        1395 - offcgalpngtest_issue1105b (Failed)
        1416 - dxfpngtest_projection-tests (Failed)
        1483 - svgpngtest_projection-tests (Failed)
        1579 - openscad-viewoptions-wireframe_view-options-tests (Failed)
        1580 - openscad-viewoptions-crosshairs_view-options-tests (Failed)
```

Full report attached:
[linux_x86_64_nvidia-geforce-gtx-1660_EXPENSIVE_default_report.html.zip](https://github.com/openscad/openscad/files/10329590/linux_x86_64_nvidia-geforce-gtx-1660_EXPENSIVE_default_report.html.zip)

I also tried testing with `CGAL_CHECK_EXACTNESS` defined, which caused no additional test failures compared to without:
0 test failures on default test config
43 failures for `ctest -C All`

thehans, 2023-01-02:
I wasn't sure if FPU rounding mode setting could get mixed up by having multiple threads and processes running concurrently, so I just ran `ctest` with no parallel jobs, and got the same results of 111 failed tests.  
So concurrency does not seem to be the cause here.

kintel, 2023-01-02:
Just to be certain: Do we know if CGAL's own tests pass with CGAL_CHECK_EXPENSIVE?

thehans, 2023-01-02:
No idea.  I couldn't figure out how to build & test CGAL last time I looked.