openscad/openscad #4500: Rendering Polygon Mesh using CGAL failed

pennywisehb, 2023-01-14:
I´m trying to expand an external stl-file, checked with Windows11 3D-Builder. 
Building the pure stl-file works fine, but adding a cube() as extension failed. My OpenSCAD 2021.01 code is:

import("D:\\Raspi\\Projekte\\PIR\\Support_mod.stl", convexity=3);
rotate([90,0,0]) translate([47, 0, -30])
    cube([10, 2, 30]);
//rotate([90,0,0]) translate([47, 38, -30])
//    cube([10, 2, 30]);
//    
//rotate([0,0,180]) translate([-58, -30, 0])
//    cube([10, 2, 40]);

[image]

jordanbrown0, 2023-01-14:
It is almost certain that the STL has some imperfection, some place where it is missing a triangle or intersects itself.

These do not show when you include the STL alone, because including the STL alone does not involve CGAL, the library that does Boolean geometry for us.  As soon as you include an additional object, CGAL gets involved and notices the problem.

Unfortunately, finding those imperfections is painful.  The first thing to do is View/ Thrown Together; look for anything purple.

If you attach the STL or supply a URL for it, we may be able to get you some help in finding and repairing the problem.

kintel, 2024-04-01:
Closing as we cannot diagnose this without the STL. It's likely a duplicate of similar issues.
A lot of similar issues work fine when using the Manifold backend.

closed 2024-04-01 by kintel