openscad/openscad #4505: App icon is incorrect under KDE Plasma Wayland session

karnhack, 2023-01-16:
Hi
I have OpenSCAD Version: 2021.01 on Arch Linux running KDE Plasma Wayland session.
Window decoration, panel show the Wayland generic icon instead OpenScad icon.

[image]

[image]

thehans, 2023-01-16:
Your "correct icon" image is nothing like the default openscad icon, so I don't think we would have any control over that.
This looks like more of an issue for the Arch package maintainers.

t-paul, 2023-01-16:
I'm seeing the issue on Debian too. Running the same app via X11 on Wayland shows the custom icon. Running plain Wayland it's just going to the default. This seems to be related to how the application name is defined.

karnhack, 2023-01-16:
> Your "correct icon" image is nothing like the default openscad icon, so I don't think we would have any control over that.
> This looks like more of an issue for the Arch package maintainers.

The icon is "correct" in relation to my icon theme (sweet). The problem also remains with the classic icons

[image]

JoKalliauer, 2023-04-26:
I have the same problem with [evince](https://wiki.gnome.org/Apps/Evince) on `Fedora` with `KDE Plasma` using `Wayland`, It might be a gtk-issue see https://gitlab.gnome.org/GNOME/gtk/-/issues/2034 and https://github.com/picoe/Eto/issues/2255 and https://bugs.kde.org/show_bug.cgi?id=409967

I do not have this issue with `Gnome` nor with `X11` only the combination of `KDE Plasma` and `Wayland` seems to cause this problem.

JKearnsl, 2023-08-18:
I have the same problem. I am using **PyQT6** and **python311**.

When I boot from **KDE** 5.27.7 (**FW**: 5.108.0 | **QT**: 5.15.10 | **Core**: 6.4.10-zen2-1-zen (64-bit)) using **Wayland**, my app icon disappears:

[image]

When I boot from **X11** everything works fine:

[image]

parkerlreed, 2023-09-09:
Yes so they are basically saying we are not going to support the old way of setting the icon all developers have to name their desktop file exactly like this and specify the icon for it to work. 

Applications that are working on Wayland right now are just because of QT and not because of the developers specifically tailoring for Wayland. It's unreasonable to expect everybody to follow the KDE guidelines just to get a window icon working.

t-paul, 2023-09-09:
There's not much point discussing general KDE topics here. Also it's not a KDE-only issue, I do see the problem with Gnome-Shell + Wayland too and not just for OpenSCAD, but for example Quassel also lost it's icon (and that's org.kde.Quassel!).

The problem I see is that it's really hard to find a definitive answer on what the correct way of naming in all the needed places is and what depends on what.

I don't have a huge issue with changing the naming in the AppData but I would have an issue if that's incompatible with X11 based systems.

So if anyone can find some clear documentation on what pieces need to play together (which files in what directories and what specific properties need to match the names defined in the Qt code used by OpenSCAD), *please* post pointers here. I failed so far to really get a full picture.

For general KDE/Qt discussion, please find some place the respective developers have a chance of seeing it.

closed 2023-12-03 by t-paul