openscad/openscad #4511: Feature Request: Preference to disable center-click–Paste

BarryTice, 2023-01-23:
I'm finding that in the process of scrolling through the code editor, I'm occasionally pasting whatever happens to be on my clipboard in the middle of the text, typically without noticing, by pressing that center mouse button (which you hold down to scroll if using the TrackPoint).

I don't see any option for turning this pasting feature off, but it would certainly make my life easier if I could do so.

I am working on a ThinkPad laptop using the three "mouse" buttons below the space bar along with the TrackPoint "nubby" mouse, and with the touch pad disabled due to issues with my thumb pads inadvertently clicking where I don't expect.

**Details:**
Version: 2021.01
OS: Ubuntu 20.04.05 (no special TrackPoint configurations to enable center-button scrolling)
Laptop: Lenovo Thinkpad T570

DarwinSurvivor, 2023-02-23:
The middle-click clipboard is a feature the operating system, similar to ctrl+v. You should disable this in your window manager. For example, if you're using gnome, I believe there's a option in gnome-tweak to disable this feature.

jordanbrown0, 2023-02-23:
Generally it is the application that maps particular events (like middle-click) to particular operations (like paste).

The OS may have conventions, and may have libraries that help you to implement those conventions, but usually it is up to the application to conform.

I don't know if I've found the right version, but it looks to me like this particular action is coming from QScintilla:
https://github.com/opencor/qscintilla/blob/d25a05d17ae952ef0bb06e9df60df15ee1b1b01b/src/qsciscintillabase.cpp#L587
```
    else if (e->button() == Qt::MiddleButton)
    {
[...]
            sci->pasteFromClipboard(QClipboard::Selection);
```

BarryTice, 2023-02-24:
Thank you for your attention!
I have been using Ubuntu since 08.04 and somehow the only place I have ever found this to be a problem is in OpenSCAD. Yet, as I'm typing this (into Firefox) and experiment with it, I see that it pastes here, too. The only other place I have ever noticed the behavior is in a terminal window, where it can be very helpful (as Ctrl-V doesn't work).
Today I learned.
Still, if it _is_ an application-configurable option, having the opportunity to turn it off would be greatly appreciated. For reasons I haven't figured out, I end up pasting code snippits into the middle of long code blocks **a lot** while editing in OpenSCAD — and that doesn't seem to happen to me anywhere else.
Thanks!

jordanbrown0, 2023-02-24:
Unfortunately, while it's an application-**implemented** behavior, that doesn't necessarily mean that it's configurable.  It's down in one of the libraries that we use, one of the ones that implements the text editor, and it doesn't immediately look like the text editor offers any way to control it.  But I can't say that I'm deeply familiar with the interface to the text editor.