openscad/openscad #4512: stopped working with error while loading libraries

coolian1337, 2023-01-24:
Hello,
I am on Artix linux and installed openscad through the AUR a couple days ago. and today after i updated my system i cant open openscad anymore with the error "openscad: error while loading shared libraries: libboost_filesystem.so.1.80.0: cannot open shared object file: No such file or directory" i tried to run sudo ldconfig and then reinstalling (with rebuild) openscad but that didnt work.

what can i try to fix this. i also tried to reinstall boost

Scopeuk, 2023-01-24:
my best guess would be that a system update has migrated you from boost 1.80 to boost 1.81 https://www.boost.org/users/history/ (latest version) and it has removed the older package which openscad was built against, a rebuild should fix that unless you system has a mixture of 1.80 headers and 1.81 libraries. I'm not familiar with Artix  to advise much more

coolian1337, 2023-01-24:
i tried it again, noticed that yay didnt actually finish installing openscad because there was already an openscad executable there, removed it, again, works