openscad/openscad #4514: Exported SVG does not import in the same place

jordanbrown0, 2023-01-27:
```
translate([10,10]) square(10);
```
Export that as SVG, then
```
import("svg export.svg");
```

Observe that the square that was at [10,10] is now at [0,40].

nophead, 2023-01-27:
I think the export looks correct and the import is broken. But web browsers show it at the origin, so who knows.

bdad2, 2023-01-31:
It would be fine if ALL svg imports set the x to 0 and y to the height. e.g. original in svg code :
<svg width="169mm" height="156mm" viewBox="-53 74 169 156" 
becomes:
<svg width="169mm" height="156mm" viewBox="0 156 169 156"

It will put those generated by openscad back into the correct position, and externally generated svg's will be visible in the original axis position, afaik. In any case if importing an externally generated svg, you will want to move it, most likely.

jordanbrown0, 2023-01-31:
With respect to *importing* an externally generated SVG, indeed it might be desirable for it to be moved, but what is critical for OpenSCAD processing is that the behavior be readily *predictable*.  A user should be able to look at the SVG and easily predict where it will end up in an OpenSCAD model.  They can then translate it from there as desired.  It would be bad, for instance, to unconditionally center the SVG, or unconditionally move it so that its bounding box starts at [0,0], because then changes to the perimeter of the drawing could cause it to shift in OpenSCAD-space, which is highly undesirable if it needs to fit with other components.  The drawing as represented in OpenSCAD space should move if and only if the drawing is moved in SVG space.

jordanbrown0, 2023-01-31:
Background:  in SVG, +Y goes down from the origin.  The idea is that the origin is at the top left corner of the page, and coordinates on the page are all positive.

In Inkscape, I created a drawing with a 50 wide 30 tall box, with its upper left corner 10 down from the top of the page and 30 right from the left edge of the page.

[image]

In SVG coordinates, that's a box with its origin at [30,10], extending down and to the right to [80,40].

Importing that drawing into OpenSCAD *looks* sensible.  It appears in OpenSCAD as if the page is in +XY with its bottom left corner at [0,0]:
[image]
(black 210x297 rectangle added for context.)

OpenSCAD seems to be negating the Y coordinate (so that +Y is up), and translating +Y by the height of the page - that is, by the height of the viewbox.

The other sensible import into OpenSCAD would be to have the page be in +X/-Y, with its top left corner at [0,0].  In a way that would be a better representation of the SVG, since native SVG has the origin at the top left, but it seems less intuitively obvious.  An import like that would negate Y without translating.

Exporting that box from OpenSCAD, on the other hand, yields a box that the Inkscape user interface shows at the origin.  (Either at the top left or bottom left corner, depending on the origin mode selected in Inkscape.)

Examining the SVG, the big difference seems to be that while the original SVG said that the viewbox (canvas? page?) was 210x297 at [0,0], the one exported from OpenSCAD has the viewbox being 50x30, at [30,-287].  OpenSCAD appears to be both (a) narrowing the viewbox to the model, and (b) negating Y so that +Y is down. It is not translating, per se; the box is still above and to the right of the coordinate system origin, but is at the viewbox origin.

That's also not a silly representation, at least with respect to the box.  It is analogous to the second import option above, where the page is in OpenSCAD's +X/-Y.  The box is appropriately above and to the right of the origin, well off the page.

So, first conclusion:  import and export are inconsistent about whether the page should be in OpenSCAD's +XY (the usual work area in OpenSCAD), or in OpenSCAD's +X/-Y (keeping the origin point at the same place it is in the SVG coordinate system).  Import has the page in OpenSCAD's +XY, while export puts it in OpenSCAD's -XY.

Then there is the question of how the viewbox comes into the picture.

(I've done less analysis here and my comments are tentative.)

Import appears to map the lower-left corner of the viewbox to the OpenSCAD origin.  Effectively it treats the viewbox origin, not the SVG coordinate system origin, as the origin for the drawing.  This means that moving the viewbox in the SVG coordinate space moves the objects in OpenSCAD space.

This combination of effects explains key points of the export-import mismatch:  export puts the viewbox origin at the upper left corner of the OpenSCAD model and sizes the viewbox to fit the model, but leaves the model at its original location (with Y negated).  Import uses the viewbox origin as the OpenSCAD origin, so the model starts at OpenSCAD [0,0] and extends into +X/-Y.

Questions:
* Should an SVG object that is in SVG's +XY, down and to the right of SVG's origin, import into OpenSCAD's +X/-Y, or (assuming that it fits onto the page) into OpenSCAD's +XY?
* Should an import be relative to the viewbox, using the viewbox to establish the origin of the resulting OpenSCAD model, or should the import be relative to the coordinate system's origin?
* Are those two questions inextricably tied together, since putting the drawing into OpenSCAD's +XY necessarily translates by the viewbox's height?
* Should export move an OpenSCAD model that is in OpenSCAD's +XY down below the X axis into SVG's +XY, or should it leave it in SVG's +X/-Y?
* Should export wrap the viewbox tightly around the model, or should it ensure that one corner of the viewbox is at the OpenSCAD origin?
* Will Polly escape from the plummeting airplane in time to save Reginald from the oncoming train?
For the answers to these questions and others, stay tuned for the next exciting episode of As the Model Rotates!

bdad2, 2023-02-04:
difference(){
    square(100,true);
        translate([-40,-40])square(30);
        translate([-40,30])square(15);
        translate([30,-25])circle(d=20);
        translate([20,0])rotate([0,0,45])square(30);
}

Is  the sort of shape I used for testing openscad generation of svg's. You can see if it is inverted, etc. I don't use inkscape, but maybe import the svg that the above creates, see if it gets inverted/whatever, re-export etc.

I use svg as a simple way of moving 2d drawings between programs, and as it's plain text, it is easily manipulated to add extra points, etc. dxf is pretty useless, many programs do not specify the version, and autodesk has de-emphasized it, anyway. The svg origin location was used for convenience, with a mindset used to the days of line printers and raster displays. I guess we have to live with it.
openscad, Irfanview, firefox, most likely others, can display (or not) the same svg in different positions, so much for the 's' part.

odisseus, 2026-09-06:
The fact that exporting a shape and importing it back does not yield the original shape is a big gotcha, and trying to align the shapes manually has cost me quite a few hours of work. I don't care too much about the exact way viewboxes are handled, but as a user of OpenSCAD I **need** the ability to export a shape and reliably import it back.

As an aside, I think that OpenSCAD should **not** invert the Y coordinate when importing SVG files. I understand the desire to make it look the same as in Inkscape or Firefox (i.e. not mirrored), but sacrificing correctness to hide complexity is the opposite of being user-friendly.