openscad/openscad #4517: OpenSCAD Freezes on macOS 12.6.3

coogle, 2023-02-04:
I'm having a problem where the following script freezes on macOS 12.6.3. It's entirely possible I've got a bug or issue somewhere in this code (I have another version that works) but it isn't obvious what's wrong since the app beachballs on me and I have to force quit. Happy to provide more code if needed.

```
use <shared.scad>
use <threads.scad>

// The size type of the container
$containerSize = "small"; // [small, medium, tall]

// The radius of the curved top of the container
$curveRadius = 15; // [0.0:0.1:15.0]

// The wall thickness
$wallThickness = 1.5; // [1.0:0.1:5.0]

// The height of the screw portion of the container
$screwHeight = 8.0; // [8.0:0.1:20.0]

// What to render
$renderPart = "all"; // [all]

if($containerSize == "small")
{
    $containerWidth = 54;
    $containerHeight = 40;
}
else if($containerSize == "medium")
{
    $containerWidth = 84;
    $containerHeight = 62;
}
else if($containerSize == "tall")
{
    $containerWidth = 84;
    $containerHeight = 142;
}
else
{
    $containerWidth = 54;
    $containerHeight = 40;
}

$fs = 0.15;

if(($renderPart == "all" || ($renderPart == "container")))
{
    color("gray")
    {
        container($containerWidth, $containerHeight, $wallThickness, $curveRadius, $screwHeight);
    }
}

module container(containerWidth, containerHeight, wallThickness, curveRadius, screwHeight)
{
    $_threadDiameter = containerWidth - 20;

    union()
    {
        // Build the container part
        difference()
        {
            roundedcube(
                size = [containerWidth, containerWidth, containerHeight],
                radius = curveRadius,
                center = true, 
                apply_to = "zmax"
            );  
            
            translate([0, 0, wallThickness])
                cylinder(r = $_threadDiameter / 2, h = containerHeight, center = true);

            roundedcube(
                size = [containerWidth - wallThickness, containerWidth - wallThickness, containerHeight - wallThickness],
                radius = curveRadius,
                center = true, 
                apply_to = "zmax"
            );  
        }

        // Build the screw-top part
        difference()
        {   
            union() {
                translate([0, 0, (containerHeight / 2) - (wallThickness/2) - 1])
                    cylinder(r = ($_threadDiameter / 2), h=wallThickness);
                translate([0, 0, (containerHeight / 2) - 1])
                    ScrewThread($_threadDiameter, screwHeight + wallThickness);    
            }
            
            translate([0, 0, (containerHeight / 2) - (screwHeight/2)])
                cylinder(r = ($_threadDiameter / 2) - ThreadPitch($_threadDiameter), h = screwHeight*4, center = true);
        }
        
    }
    
}
```

kintel, 2024-04-01:
@coogle Does this still crash? If you could provide a complete example, I can try to diagnose it.

coogle, 2024-04-01:
TBH this was months ago now and I haven't circled back to it. I can provide the include files referenced (otherwise it looks like I already provided all the code in question) if that would be helpful.

kintel, 2024-04-01:
If you share the files, I can at least comment on whether the freeze is expected (e.g. processing time is expected to take really long).

coogle, 2024-04-01:
Sure thing, attached!
[supporting-code.zip](https://github.com/openscad/openscad/files/14819226/supporting-code.zip)

closed 2024-12-22 by kintel