openscad/openscad #4518: unable to type more than three digits into an input box in the customizer

krhackbarth, 2023-02-06:
If I add a line like this to the customizer sensitive area of the code:
input_value  = 0;

The customizer shows a spinner control with the value 0.  I can then type (or paste) any integer value from -999 to 999 into the control.  I can use the spinners themselves to increase the value further but this is a laborious process.  For all intents and purposes it's impossible to set the value to something like 9999.

If I use this code:
input_value = 0; //[0-10000]

I can enter values as large as 9999 into the box but at the cost of including a slider bar in the UI which is practically unusable.

I would prefer to be able to insert the first line of code above and not be restricted as to the magnitude of the integer I enter/paste into the box.

Intelligently supporting floating point values would be really nice as well - but that's a different discussion.

bitbasher, 2025-06-16:
i cannot reproduce this issue. When i define a parameter as suggesting in the OG description i get a spin-box widget that i can enter any numeric value into, negative, decimal ... the only thing i cannot enter is a science notation value, like 5e2 .. but that has never worked for customizer params.

this issue can be closed

krhackbarth, 2025-06-16:
As you get closer to release, I'll definitely be doing that.

bitbasher, 2025-06-16:
my testing is in the nightly build .. i am currently using this:

OpenSCAD Version: 2025.03.31 (git 3c1f9e779)
System information: Microsoft Windows 11 (10.0.26100) x86_64 32 CPUs 31.77 GB RAM
User Agent: OpenSCAD/2025.03.31 (git 3c1f9e779) (Microsoft Windows 11 (10.0.26100) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1