openscad/openscad #4520: Feature request: Get the dimensions of child objects

eintr, 2023-02-14:
It's nice to have a build-in function to get the x/y/z size or range of a child object. Be like:

`echo(sizeof(children(1)));`

we can get: [1,2,3] for 3D, and [1,2] for 2D, or

`echo(rangeof(children(1)));`

we can get: [[1,2],[2,3],[3,4]] for 3D, and [[1,2],[2,3]] for 2D

jordanbrown0, 2023-02-14:
PR#4478 includes features that would allow this, via the render function and geometry literals.

```
module x() {
    geometry = {{ children(); }};
    rendered = render(geometry);
    echo(size=rendered.size, min=rendered.min, max=rendered.max);
}

x() cube([1,2,3]);
```

> ECHO: size = [1, 2, 3], min = [0, 0, 0], max = [1, 2, 3]

Note, however, that it is not nearly as performance-cheap as you would think.  Calculating the bounding box of an intersection or difference requires a full render of the mesh involved.  A preview that runs in a fraction of a second could take minutes when it requires a full render of a complex part.

t-paul, 2023-02-14:
Will be covered by the "object" work referenced above.

closed 2023-02-14 by t-paul