openscad/openscad #4522: Segfault on --render with scale to 0

rcolyer, 2023-02-15:
```cube(1); scale([1, 0, 0]) cube(1);```

and then:

```openscad --render 512,512 -o test.png test.scad```

produces a segmentation fault in the master branch, but not for the last release.

jordanbrown0, 2023-02-15:
Also for F6 in the GUI in my experimental build.  (Which does include my experimental stuff, but shouldn't have touched this.)

rcolyer, 2023-02-15:
It did not work for me in the gui, but I realize now this is because of fastcsg being enabled.  Notably:

```openscad --enable fast-csg --render -o test.png test.scad```

Does NOT cause the segfault.  So it is specifically a regression in the non-fast-csg path.

ochafik, 2023-02-23:
This crashes inside CGAL' Nef constructor from Polyhedron_3 ([called by createNefPolyhedronFromPolySet](https://github.com/openscad/openscad/blob/master/src/geometry/cgal/cgalutils.cc#L59)), it's choking on a null halfedge (this [commented out assert](https://github.com/CGAL/cgal/blob/master/Nef_3/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h#L222) would have failed and avoided a crash if caught by us) but doesn't seem to be a regression in CGAL (built against various CGAL versions with the following script).

The minimal repro is `echo "polyhedron([[0, 0, 0]], [[0, 0, 0]]);" | openscad - -o out.nef3` (or `scale(0) cube();`)

I'm wondering if the hull changes I did in https://github.com/openscad/openscad/commit/2b11508f2b1f163827db3de0189ae8975c75f6b1 messed up the case of degenerate polyhedra by avoiding the output of duplicate vertices too aggressively maybe.

Need more digging, possible fix in CGAL or maybe we could test for degenerate polyhedron for early exit instead of just [#points <= 3 here](https://github.com/openscad/openscad/blob/master/src/geometry/cgal/cgalutils.cc#L52).

```bash
#!/bin/bash
set -eu

for v in 5.5.1 5.5 5.4.3 5.4.2 5.4.1 5.4 5.3.2 5.3.1 5.3 ; do
  if [[ ! -d ../CGAL-${v} ]]; then
    ( cd .. && \
      curl -L --insecure https://github.com/CGAL/cgal/releases/download/v${v}/CGAL-${v}-library.tar.xz --output CGAL-${v}.tar.xz && \
      xz -f -d CGAL-${v}.tar.xz && \
      tar xf CGAL-${v}.tar )
  fi
  ( mkdir -p buildDebug-${v} && \
    cd buildDebug-${v} && \
    cmake -DEXPERIMENTAL=1 -DCMAKE_BUILD_TYPE=Debug -DCGAL_ROOT=../../CGAL-${v} .. && \
    make -j12 )
done
```

Also: tried to bissect but couldn't find a good commit that builds easily on my mac yet (earliest bad commit I managed to build is https://github.com/openscad/openscad/commit/d9acaca55bb55f738f6178354cd986f42e51edf5 back in May'22)

PR #4530 (2023-02-23): Avoid duplicate points in PolySet::quantizeVertices's output vector
This should fix #4522 (degenerate geometry now yields 3 points or less after hulling and isn't hitting the Nef polyhedron's ctor).

https://github.com/openscad/openscad/pull/4170 sneaked the list of points out of the quantization routine, but the logic to avoid dupes was bogus (it only avoided them within the same polygon, so if all polygons had dupes these were still let through)
commit 648d74d8 (2023-02-23): Avoid duplicate points in PolySet::quantizeVertices's output vector (fixes #4522)
closed 2023-02-23 by t-paul