openscad/openscad #4523: Edges - Bug or feature ?

gsohler, 2023-02-15:
Hi, 

I have put two simple cubes next to each other and pressed F6 for CGAL rendering.
The result looks like:

https://imgpile.com/i/dB6N9h

If you closely watch the picture, you will see a yellow square in the top of the 1st cube and it moves with the corner in either
viewport.

* It appears to be a selected corner, but its only visible when button "Show Edges" is active
* its only visible during F6 and its only visible when an union is being performed

If its a feature: How can i select the other corners ?
And why does it enable corners instead of edges ?
and why is only visible with F6 ?
And How to select the other edges ?

Any Answer makes me happy :)

nophead, 2023-02-15:
Looks like a bug. I get this:

[image]

With OpenSCAD 2021.10.19.ci9029

gsohler, 2023-02-15:
very intersting. my picture is from OpenSCAD 2022.12.06.ai12948(git 972d18464), believe i got it with yum update
same result with my self compiled version
i am wondering why your faces are green with F6 aka Cgal. probably a setting

does anybody remember such a bug was fixed some time in between ?

nophead, 2023-02-15:
Yes the green instead of yellow is just the colour scheme setting. Show edges normally shows all the convex edges. It does have a bug where the concave edges are rendered inside the object. So it looks like a regression. I can't run more recent versions.

jordanbrown0, 2023-02-15:
First... it's definitely not "selected".  OpenSCAD doesn't have such a concept.

It's hard to be sure in a screen shot in the default color scheme, but what looks wrong to me is not the presence of the small cube-oid in that corner, but rather the *absence* of similar cuboids at other corners.  @gsohler, you might try selecting a different color scheme to see if it looks different.

@nophead's screen shot shows what I expect from View Edges - cubes at the vertices and lines along the edges.  I get that from both 2021.01 and a relatively recent bleeding edge build, on Windows.

It appears that you get those things *only* if CGAL gets involved, which means only if you F6 and only if there's more than one object.

gsohler, 2023-02-15:
Hi Jordan,
I tried another preference where the faces are green, and i could still
only see the yellow rectangle i the same corner,
only with the exact testcase and only when pressing F6.
( dont have a openscad development setup on this computer, so its not my
code change here)

Now I am at home and i use another computer where i installed OpenSCAD
2022.12.06.ai12948(git 972d18464)
i turned the faces green already, but i cannot see the corner squares
anywhere.
if you can point me to the line in code which  create the rectangle, i can
debug why i dont see the rectangle.

[image: image.png]

jordanbrown0, 2023-02-15:
I don't know where in the program the "show edges" feature is implemented.

However, if you're not seeing edges and vertices marked, I would suggest confirming that  you have two objects, have done a full F6 render, and have selected "view edges".  It might (or might not) be important what order you do those things in.

gsohler, 2023-02-16:
I Just confirmed again on a virgin openscad installation(yum install) , that i have pressed "F6" and i have ticked the checkbox "Show Edges"  , but only one of the 16 corners were annotated with a yellow rectangle .

Jordan, after investigating, it turns out, the code to display the rectangle is  somewhere deep in CGALRenderer.cc, CGALRenderer::draw()

While all the corners show up correctly in your place you did not mention that you have vertex-object-renderer enabled in your experimental Features.  When I enable this feature in my place, too all corners are shown correctly.

Suppose you can agree to my theory

However this is not the default after installation and it appears that the experimental Feature works better than the default.
So is it better to make this feature default or rather fix the error in the default configuration ?

nophead, 2023-02-16:
For me it works without vertex-object-renderer enabled and show edges pre-dates that addition. Perhaps it is a graphic driver bug on your system.

gsohler, 2023-02-16:
i tested it in linux FC37 inside VirtualBox and in Linux FC37 plain, but (unkonwn/old Graphics card)
both have different bogus effects

nophead, 2023-02-16:
I am on Win7 on an HP laptop but the show edges feature has always worked for me on all my machines since OpenSCAD was first released.

gsohler, 2023-02-16:
This is a little bit desperate for me now.

If anybody sees same effect like me,  please tell me
If anybody know the location of the the actual gl* code for the yellow
square, please tell me

In case there are no answers, I will probably have to outsource the bug
report to my-computers "repository" :)

jordanbrown0, 2023-02-16:
Sorry, no, I see the same behavior in 2021.01 and in the bleeding edge, and I don't have the vertex-object stuff enabled.  I have no idea why you see something different.

[image]

jordanbrown0, 2023-02-17:
I did not test Linux.  I don't routinely have a Linux system running.

And yes, it would be good if it worked in all F6 cases.
In F5/preview/OpenCSG cases we don't know where the vertices and edges are.

gsohler, 2023-02-20:
Right now it really appears, that the "bug" is there in openscad for Linux, but not in openscad for Windows.
Anybody else which can confirm the  error exists in openscad for linux ?

nophead, 2023-02-20:
It looks more like a graphics driver bug to me than a difference in OSes.

gsohler, 2023-02-20:
nohead, which driver do you exactly refer to ?
How can i find the version of this driver in FC37 ?
(and how could i update in case its not the latest ?)

nophead, 2023-02-20:
It's the OpenGL driver for the graphics card that your computer has. I am not familiar enough with Linux to know what it is called or where it lives. I don't have access to a Linux machine in the winter.

rcolyer, 2023-03-22:
I didn't realize others were seeing squares at the vertices with "show edges" enabled.  (It doesn't say anything about vertices in the name.)  But I do not see this under Linux for any of the vertices.  (Also those vertex square are not very pretty.)  Perhaps it has something to do with platform-specific differences in the widths of these objects?  I note that the edge highlights I get are very thin.  Probably too thin if the goal is high visibility, but I have never complained because I didn't want thick crayon lines on my models.  ;)  But if the edges are thin, perhaps vertex points are as well, and consequently just points?

[image]

jordanbrown0, 2023-03-22:
Preview with show-edges gives thin lines along the edges and nothing at the corners, like you show.

Render with show-edges *sometimes* gives no edge markings and *sometimes* gives thicker lines along edges, with blocks at corners.  Wireframe sometimes works and sometimes doesn't, and correlates with the show-edges behavior.

I have not been able to characterize when show-edges and wireframe work and don't work.

gsohler, 2023-03-22:
Thank you for all your answers

I am not interested in getting this resolved anymore as it's apparently
something very specific to my setup.
i just wanted to point out the issue, but now I found out the error must be
on my side.
I am happy to  close  the issue.

kintel, 2023-04-09:
I agree that this smells like a driver issue. For posterity: This is how the vertices are rendered in regular vs. VBO mode:
https://github.com/openscad/openscad/blob/102118fbf3b9d917d58c899af11b15db4249c1ea/src/ext/CGAL/OGL_helper.h#L389
https://github.com/openscad/openscad/blob/102118fbf3b9d917d58c899af11b15db4249c1ea/src/glview/cgal/CGAL_OGL_VBO_helper.h#L270