openscad/openscad #4525: Problems in the preview

kberkut, 2023-02-20:
Problems in the preview when cutting an object from linear extrusion.
`$fn = $preview ? 100 : 100; 
module move(x=0,y=0,z=0,rx=0,ry=0,rz=0) 
{translate([x,y,z]) rotate([rx,ry,rz]) children();}
difference(){
linear_extrude(10) import("/work/Temp/temp.dxf");
  # move(y=-25,z=-1) cube([100,100,20]);
}`

Rename the attached file from txt to dxf
[temp.txt](https://github.com/openscad/openscad/files/10783095/temp.txt)

OpenSCAD Version: 2021.01
System information: Linux 6.1.12-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 14 Feb 2023 22:08:08 +0000 x86_64 Arch Linux 8 CPUs 9.66 GB RAM
User Agent: OpenSCAD/2021.01 (Linux x86_64; Arch Linux)
Compiler: GCC "12.2.1 20230111" 64bit
MinGW build: No
Debug build: Yes
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.15.8
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.74.5
lodepng version: 20180910
libzip version: 1.9.2
fontconfig version: 2.14.1 (runtime: 2.14.2)
freetype version: 2.12.1 (runtime: 2.13.0)
harfbuzz version: 6.0.0 (runtime: 7.0.0)
cairo version: 1.17.6 (runtime: 1.17.8)
lib3mf version: 1.8.1

t-paul, 2023-02-20:
Looks like a convexity issue, see [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?)

kintel, 2023-04-09:
Definitely convexity issue; use `linear_extrude(10, convexity=2)`

closed 2023-04-09 by kintel