openscad/openscad #4527: Trying to compile with "EXPERIMENTAL"

mconsidine, 2023-02-22:
Hi,
I am on Linux Mint 21 and am trying to compile the latest codebase with -DEXPERIMENTAL=ON

During the cmake execution I see the following:
`-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.6", minimum required is "3.4") 
-- Preparing image_compare.py for test suite image comparison
-- venv not found
-- Setting up testing venv for image comparison
-- Found Python3: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 
/home/matt/Downloads/openscad-build/tests/venv/bin/python: No module named ensurepip
/home/matt/Downloads/openscad-build/tests/venv/bin/python: No module named pip
Traceback (most recent call last):
  File "/home/matt/Downloads/openscad-source/tests/image_compare.py", line 3, in <module>
    from PIL import Image
ModuleNotFoundError: No module named 'PIL'
CMake Warning at tests/CMakeLists.txt:100 (message):
  Failed to setup the test suite venv for
  /home/matt/Downloads/openscad-build/tests/venv/bin/python See
  doc/testing.txt for dependency information.
`
From a command line, I can start python and import PIL.  I can also run pip

The only other warning I see is ahead of this, namely,
`- OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so;/usr/lib/x86_64-linux-gnu/libGLX.so;/usr/lib/x86_64-linux-gnu/libGLU.so
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/CGAL/CGALConfig.cmake:92 (message):
  CGAL_DATA_DIR cannot be deduced, set the variable CGAL_DATA_DIR to set the
  default value of CGAL::data_file_path()
Call Stack (most recent call first):
  CMakeLists.txt:283 (find_package)

-- Using header-only CGAL
`
A google and github search don't seem to yield any answers to how fix the circumstances generating the warnings.

Can anyone provide any pointers, advice or solutions?
TIA,
mconsidine

thehans, 2023-02-23:
You are missing a dependency for the recently updated python-based image comparison used by our test suite

You need `python3-venv` and `python3-pil` installed for those, as documented here:
https://github.com/openscad/openscad/blob/master/doc/testing.txt

If you aren't interested in running tests, then I think adding cmake argument: `-DENABLE_TESTS=OFF` should avoid that dependency check (I haven't tried it yet).
We should maybe update the build instructions in the main README with a note about this.  Or maybe change the default value of `ENABLE_TESTS` to `OFF`

mconsidine, 2023-02-23:
@thehans:. Thank you very much!  I'll work on that tomorrow.  ( I had looked for the testing.txt doc, but was searching in completely the wrong place).

Much obliged,
mconsidine

mconsidine, 2023-02-23:
Thanks again for the pointer.  The python pieces were in fact missing.  I get errors with it trying to find shader source files, but that's a different problem.
Closing this ...

closed 2023-02-23 by mconsidine