openscad/openscad #4529: if..else..if block generates empty geometry

DarwinSurvivor, 2023-02-23:
The behaviour between "if" and "if else if" is inconsistent in whether it generates nothing or empty geometry.

The following generates nothing
```
if(false)
  cube();
```

The following generates empty geometry (object of zero volume)
```
if(false)
  cube();
else if (false)
  cube();
```

This become very evident when wrapped in an intersection with another object.

The following renders a cylinder since it's an intersection of 1 object (the cylinder)
```
intersection() {
  cylinder();
  if (false)
    cube();
}
```

The following renders nothing because it's an intersection of 2 objects (the cylinder and the zero-volume object generated by the if..else..if block)
```
intersection() {
  cylinder();
  if (false)
    cube();
  else if (false)
    cube();
}
```

I believe an if..else..if block with all false conditions should create nothing so as to be consistent with a false if statement.

jordanbrown0, 2023-02-23:
An `if` that evaluates a block generates geometry, a `group` node containing the results of evaluating the block.

```
if (false) { }
```
does not evaluate a block and so generates nothing.
```
if (true) { }
```
evaluates the block and so generates a group node, and the block generates nothing.
```
if (false) { }
else { }
```
evaluates the `else` block and so generates a group node, and the block generates nothing.

```
if (false) { }
else {
    if (false) { }
}
```
evaluates the `else` block, and so generates a `group` node, and then the second `if` generates nothing.

---

Net, it's consistent:  if the `if` evaluates a block, it generates geometry.

Is it ideal?  Maybe, maybe not, but it's been that way for a long time so we're stuck with it.

Note that if you want it to consistently generate geometry, you can always include an empty `else` block.

One case where this causes a problem is an else-if chain
```
if (c1) { ... }
else if (c2) { ... }
else if (c3) { ... }
```
where you might think that you would get nothing if none of the conditions are true, but since it's really parsed like so:
```
if (c1) { ... }
else {
    if (c2) { ... }
    else {
        if (c3) { ... }
    }
}
```
the `else` blocks generate `group` nodes.

We could introduce an explicit `elseif`:
```
if (c1) { ... }
elseif (c2) { ... }
elseif (c3) { ... }
```
which would generate a `group` node only if it evaluates a block.

jordanbrown0, 2023-02-23:
Workaround:
```
if (c1) { ... }
if (!c1 && c2) { ... }
if (!c1 && !c2 && c3) { ... }
```

jordanbrown0, 2023-02-23:
We could probably also parse `else if` specially, but changing that behavior would at least in theory be incompatible.