openscad/openscad #4534: fs::is_directory fails in nigthly v

kakila, 2023-03-01:
In nightly version "OpenSCAD version 2022.12.07" from snap store in Ubuntu 22.04
I get failures when generating stl from the command line

```
$ openscad-nightly -o/tmp/stl/housing_DesignTyp4.stl -D component="housing"  wabe.scad
'/tmp/stl' is not a directory for output file /tmp/stl/housing_DesignTyp4.stl - Skipping
```
The failure is given by fs::is_directory

Scopeuk, 2023-03-01:
I believe snap is sandboxing it's temp folder (so what the application inside snap sees as /tmp is specific to that application and not the real /tmp (https://forum.snapcraft.io/t/accessing-tmp-from-snaps/22384) . I believe by default you have access to things under ~ but I am not certain.

kakila, 2023-08-22:
Thanks for the suggestion. I created `~/tmp` and tried
```
$ openscad-nightly -o ~/tmp/housing_DesignTyp4.stl -D component="housing"  wabe.scad
'/tmp/stl' is not a directory for output file /tmp/stl/housing_DesignTyp4.stl - Skipping
```
This is with 
```
$ openscad-nightly --version
OpenSCAD version 2023.08.20
$ ls -al $(which openscad-nightly)
lrwxrwxrwx 1 root root 13 Aug 22 09:43 /snap/bin/openscad-nightly -> /usr/bin/snap
```
The command works with the apt default openscad
```
$ openscad --version
OpenSCAD version 2021.01
$ ls -al $(which openscad)
-rwxr-xr-x 1 root root 8011024 Mär 17  2022 /usr/bin/openscad
```

Any suggestion to solve the issue?

Scopeuk, 2023-08-22:
the output on the first command looks like it was openscad-nightly -o ~/tmp/stl/housing_DesignTyp4.stl -D component="housing"  wabe.scad inline with your first example but the folder ~/tmp/stl did not exist. if it was run with the command as show than is very strange and looks like something added /stl/ to the path for some reason. could you confirm if this is the case?

kakila, 2023-08-22:
my bad, once I create the folder it works. Sorry for that.
Is there a way to find the temporary folder of an snap app? (slightly off topic, but maybe you know. I have a question here https://askubuntu.com/questions/1483246/find-snap-app-temporary-folder-programatically)

kakila, 2023-08-22:
I think the error message is misleading. Is there a way to check if opens and has  write access to the directory? Then inform the user respectively.
If I understand the intention of the code, it is trying to check if the directory exists. In this case it does, but the app cannot write to it.
Maybe 'cannot write to folder' is a better error message, instead of implying it doesn't exists.