openscad/openscad #4544: Crash during compile of qrc_common.cpp

threed-factory-store, 2023-03-10:
I'm on Ubuntu 22.10.

I cloned the repository, ran uni-get-dependencies.sh and check-dependencies.sh and everything came back OK.

Generated the makefile, ran make -j, and it crashed on qrc_common.cpp.  Took the terminal window with it.

t-paul, 2023-03-10:
Well, don't do that (`make -j`) unless you have unlimited memory. Try `make -j2` and increase if you have enough memory.

closed 2023-03-10 by t-paul

threed-factory-store, 2023-03-10:
> Well, don't do that (make -j)

Then would you fix this page?
https://github.com/openscad/openscad#building-openscad

t-paul, 2023-03-10:
Thanks for the pointer. Suggesting that in the readme makes no sense.

threed-factory-store, 2023-03-10:
Thanks, and thanks for the quick response!

t-paul, 2023-03-10:
Readme is updated now.