openscad/openscad #4546: Is rendering issue or some thing else?

cidylong, 2023-03-10:
I'm trying to create a corrugated sheet mould to handle my corrugated sheet bending.

I have [OpenSCAD](https://www.openscad.org/) version 2021.01 installed on my MacOS 13.2.1 (22D68). 

I have scad code as:

$fa = 0.5; 
$fs = 0.5;

sheet_l = 150;   // length of corrugated sheet
sheet_t = 0.42;   // thickness of corrugated sheet
minus_t = 20;
block_t = 30;
wave_l=76.2;
r = 24;       // radius of the curve
a = 2*asin((wave_l/4)/r);       // angle of the curve

function isOdd(x) = (x % 2) == 1;

module curve(thicker){
    translate([-(r*cos(a/2)),0,0])rotate([0,0,-a/2])rotate_extrude(angle = a) translate([r, 0, 0]) square(size = [thicker, sheet_l], center = true);
}

/*curve(sheet_t);*/

module corrugated_sheet(thicker){
    union(){
        curve(thicker);
        for (i=[1:12]){
                if (isOdd(i)){
                    translate([0,i*wave_l/2,0])rotate([0,180,0])curve(thicker);
                    translate([0,-i*wave_l/2,0])rotate([0,180,0])curve(thicker);
                }   
                else{
                    translate([0,i*wave_l/2,0])curve(thicker);
                    translate([0,-i*wave_l/2,0])curve(thicker);
                }
            }
    }
}

/*corrugated_sheet(sheet_t);*/
module base_support(){
    difference(){
        translate([-5,0,0])cube([block_t,wave_l*12,sheet_l],true);
        translate([10,0,0])corrugated_sheet(20);
        /*translate([10,0,0])corrugated_sheet(sheet_t);*/
    }
}

base_support();

module splitToTwo(){
    difference(){
        cube([block_t*2,wave_l*12,sheet_l],true);
        corrugated_sheet(sheet_t);
    }
}

/*splitToTwo();*/

When I run [preview - F5], it display properly, As:

[image]

But, when I run [Render -F6], The rendered result is not I expected. 

[image]

As you can see, there are two rabbit ears (paper thin) popped up in the crest of corrugated.

jordanbrown0, 2023-03-11:
The program quoted does not work for me.
* It's got lines that look like so:
```
/something();/
```
which are syntactically invalid.
* If I assume those are supposed to be commented out... it's got references to a variable `iwave_l` that is not defined.
* If I assume those are really supposed to be `wave_l`... I only get two notches, not 12.

Some of this might be caused by GitHub messing with the formatting.

Please verify that the program works, then edit the original report to enclose the program in triple-backquotes so that GitHub will not mess with the formatting.

As you enter the program it will look something like so:
```
    ```
    translate(whatever) {
        something();
    }
    ```
```
and on the preview tab should look something like so:
```
translate(whatever) {
       something();
}
```

cidylong, 2023-03-11:
/something();/ is modified by github. Original code as:

    /*something();*/

it is comments

cidylong, 2023-03-11:
I will replace it by using code style:

jordanbrown0, 2023-03-11:
I looked at this a little more, and realized that a key problem was that GitHub had eaten the `*` from `i*wave_l`.  That got me the intended pattern and let me duplicate the behavior described.

There is no bug, except in that the behavior of zero-thickness (or perhaps near-zero-thickness) objects is ill-defined.

[image]

First:  The Z-fighting at the edges obscures the picture.  Make your `curve()` be a little larger in Z to eliminate it ([FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?)).  I would add `1` to the `sheet_l` in `curve()`.  That will make it clearer that the thin "ears" are there in the preview, but flickering.

[image]

Second:  note that the flickering ears are precisely at the joints between the curves that face up and the curves that face down.  Maybe add `#` in front of the first `translate` in `curve()` to see.

[image]

You're subtracting away the volume on both sides of that joint precisely enough that it's zero thickness, or very close to it.  F5 preview can't quite decide if it's there; F6 render decides that it is there.

Make sure that the two sets of curves overlap, so that there are no zero-thickness volumes.  I would add `1` to the value of `a` to make each curve just a little bigger.

[image]

cidylong, 2023-03-11:
$fa = 0.5; 
	$fs = 0.5;

	sheet_l = 150;   // length of corrugated sheet
	sheet_t = 0.42;   // thickness of corrugated sheet
	stack_step = 0.40;
	minus_t = 20;
	block_t = 30;
	wave_l=76.2;
	r = 24;       // radius of the curve
	a = 2*asin((wave_l/4)/r);       // angle of the curve

	function isOdd(x) = (x % 2) == 1;

	module curve(thicker){
	    translate([-(r*cos(a/2)),0,0])rotate([0,0,-a/2])rotate_extrude(angle = a) translate([r, 0, 0]) square(size = [thicker, sheet_l], center = true);
	}

	/*curve(sheet_t);*/

	module corrugated_sheet(thicker){
	    union(){
	        curve(thicker);
	        for (i=[1:12]){
	                if (isOdd(i)){
	                    translate([0,i*wave_l/2,0])rotate([0,180,0])curve(thicker);
	                    translate([0,-i*wave_l/2,0])rotate([0,180,0])curve(thicker);
	                }   
	                else{
	                    translate([0,i*wave_l/2,0])curve(thicker);
	                    translate([0,-i*wave_l/2,0])curve(thicker);
	                }
	            }
	    }
	}

	/*corrugated_sheet(sheet_t);*/

	module sheetStack(){
	    union(){
	        corrugated_sheet(sheet_t);
	        for(i=[1:25]){
	            translate([i*stack_step,0,0])corrugated_sheet(sheet_t);
	            translate([-i*stack_step,0,0])corrugated_sheet(sheet_t);
	        }
	    }
	}

	/*sheetStack();
	translate([60,0,0])corrugated_sheet(20);*/

	module base_support(){
	    difference(){
	        translate([-5,0,0])cube([block_t,wave_l*12,sheet_l],true);
	        translate([10,0,0])corrugated_sheet(20);
	        /*translate([10,0,0])corrugated_sheet(sheet_t);*/
	    }
	}

	base_support();

	module base_inner(){
	    difference(){
	        translate([5,0,0])cube([block_t,wave_l*12,sheet_l],true);
	        translate([-10,0,0])corrugated_sheet(20);
	        translate([0,0,-86])rotate([0,45,0])cube([block_t,wave_l*12,sheet_l],true);
	        /*translate([10,0,0])corrugated_sheet(sheet_t);*/
	    }
	}

	base_inner();

	module splitToTwo(){
	    difference(){
	        cube([block_t*2,wave_l*12,sheet_l],true);
	        corrugated_sheet(sheet_t);
	    }
	}

	/*splitToTwo();*/

cidylong, 2023-03-11:
Thank you Jordan. I will modify my code and test it again.

jordanbrown0, 2023-03-11:
Protip:  For this kind of shape, working in 2D with `polygon()` and list comprehensions, and then `linear_extrude`, can be a lot simpler and faster.

I used a sine wave rather than your connected circular arcs, because I didn't feel like spending the energy to figure out the function for the circular arcs, but this should give you the idea:
```
sheet_l = 150; // length of corrugated sheet
sheet_w = 1900;
block_t = 30;
r = 24; // radius of the curve
step = 0.01;

linear_extrude(height=sheet_l) polygon([
    [ sheet_w/2, -block_t ],
    [ -sheet_w/2, -block_t ],
    for (x=[-6:step:6]) [ x*sheet_w/12, r*sin(x*360) ]
]);
```
[image]

Substitute your own function as desired.

cidylong, 2023-03-11:
Hi Jordan.

Thank you for your effort and advice.

Actually, the corrugated sheet (Australia) definition is not a sine wave.

cidylong, 2023-03-11:
It is a little bit hard to join circular arcs to form a polygon.

So there is no way to create mesh by using linear_extrude().

If OpenSCAD supports creating circular arcs by gaven radius (r) and angle
(a) in the future, this sort of work will be very easy.

Have a good weekend!

Cidy from Australia

UBaer21, 2023-03-11:
i solved this in my lib ub.scad  .. so you can copy the code 
`linear_extrude(20)Welle(e=12,r=5,r2=8,grad=90,rand=0,ext=-10);`
[image]

jordanbrown0, 2023-03-11:
Creating polygons is just math.  Circular arcs are easy trigonometry:
```
function arc(center, r, astart, astep, astop) = [
    for (a = [astart:astep:astop]) center + [r*cos(a), r*sin(a)]
];

astart = 10;
astop = 110;
astep = 1;
r = 5;
center = [3,3];

points = [
    center,
    each arc(center, r, astart, astep, astop)
];
polygon(points);
```
You must need to connect up the right set of them, offset to the right origins.

It's figuring out the right center points that I was leaving as an exercise for the reader.

[image]

kintel, 2024-04-01:
Status as of 2024-03-24: The original example now works with Manifold, but is still broken with CGAL.