openscad/openscad #4547: Segmentation fault when generating STL file from Scad file

threed-factory-store, 2023-03-11:
When I run 
      openscad -o 
on the attached Scad file, it crashes with the following message:

1121795 Segmentation fault      (core dumped) openscad -o "${StlFile}" "${ScadFile}"

[SegFault.zip](https://github.com/openscad/openscad/files/10948647/SegFault.zip)

craftmaster1231, 2023-05-05:
Any details about version, environment?
Everything works fine in my case.
![image](https://user-images.githubusercontent.c

Just in case you need that stl file:
[stl.zip](https://github.com/openscad/openscad/files/11409965/stl.zip)
om/32911478/236565799-71aa5cc8-a486-4336-930a-895cb89bbbcf.png)

craftmaster1231, 2023-05-05:
> Any details about version, environment? Everything works fine in my case. [image]
> 
> Just in case you need that stl file: [stl.zip](https://github.com/openscad/openscad/files/11409965/stl.zip) om/32911478/236565799-71aa5cc8-a486-4336-930a-895cb89bbbcf.png)

Of course, after path updated to `use <fillets3d.scad>` .

threed-factory-store, 2023-05-06:
Ah.  So I had two versions of OpenScad installed on Ubuntu, both from the Software Center.    I was runnning version 2021.01 which is "openscad" in the terminal.  The other version is called "openscad-nightly".  If I run that version, it generates the STL file.

Thanks for your help!

closed 2023-05-06 by threed-factory-store