openscad/openscad #4558: "mesh is not closed" for some charactor in some fonts.

Blue-Beaker, 2023-03-17:
OpenSCAD Version: 2021.01-4build1 in Ubuntu 22.04 sources

When I use "Unifont" font to create texts, it previews well but always errors this when rendering:  
`ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.`  

In the font "Noto Sans CJK SC", there's also some glyphs that can't be rendered in OpenSCAD, like "螺".  

I think it's the problem that these glyphs have crossed paths so they causes crossed and non-closed meshes when extruded.  
Since we have no control over font files, I think it should be fixed from OpenSCAD side.  
A possible solution is to cut path at every cross when generating text, dividing it into individual closed paths.

thehans, 2023-03-17:
As far as I know, unifont is some form of bitmap font, and I think OpenSCAD is only capable of supporting vector based fonts.

For the glyph in Noto Sans CJK SC, could you provide sample code to reproduce the issue?  
I assume you are linear_extruding the font and then attempting CSG on it?

Clipper library, which we use for most 2D operations, has some functions which might be able to help auto-repair self-intersections, assuming that's the issue here.

jordanbrown0, 2023-03-17:
I don't immediately find a "Noto Sans CJK SC".  I did find "[Noto Sans SC](https://fonts.google.com/noto/specimen/Noto+Sans+SC)"

I duplicate the problem with Noto Sans SC:style=Bold (but not with the other variations).

```
f = "Noto Sans SC:style=Bold";
s = "\u87ba";   // "螺"
// echo(fontmetrics(font=f));
linear_extrude(height=1) text(s, font=f);
cube(0.1);
```
> ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.

jordanbrown0, 2023-03-17:
I believe the problem is a non-manifold point at the bottom center of the glyph.
[image]

```
f = "Noto Sans SC:style=Bold";
s = "\u87ba";   // "螺"
// echo(fontmetrics(font=f));
linear_extrude(height=1)
    intersection() {
        #text(s, font=f);
        translate([7.2,0])
            square([0.3,0.3]);
    }
cube(0.1);
```

Applying a tiny offset appears to work around the problem.
```
f = "Noto Sans SC:style=Bold";
s = "\u87ba";   // "螺"
linear_extrude(height=1)
    offset(0.001)
    text(s, font=f);
cube(0.1);
```

jordanbrown0, 2023-03-17:
I used an experimental geometry-as-data feature to pull this segment apart.

```
f = "Noto Sans SC:style=Bold";
s = "\u87ba";   // "螺"
poly = {{
    intersection() {
        text(s, font=f);
        translate([7.3,0])
            square([0.2,0.3]);
    }
}};

//*union() {
//    linear_extrude(height=1) poly;
//    cube(0.1);
//}

r = render(poly);
echo(r);

pts = r.objects[0].points;
for (i=[0:len(pts)-1]) {
    p = pts[i];
    translate(p) {
        text(str(i), size=0.03, halign="center", valign="center");
        color("black") circle(0.003, $fn=20);
    }
}
```
[image]
Note that points 1 and 5 are in the same place.
Here's the results from render(), reformatted for readability:
```
ECHO: {
    min : [7.3, 0, 0],
    max : [7.5, 0.3, 0],
    center : [7.4, 0.15, 0],
    size : [0.2, 0.3, 0],
    objects : [
        {
            points : [
                [7.5, 0.191371],
                [7.3611, 0.1944],    <<< point 1
                [7.44094, 0.3],
                [7.3, 0.3],
                [7.3, 0.115491],
                [7.3611, 0.1944],    <<< point 5
                [7.4431, 0],
                [7.5, 0]
            ],
            paths : [[0, 1, 2, 3, 4, 5, 6, 7]]
        }
    ]
}
```

jordanbrown0, 2023-03-17:
I duplicated the Unifont problem using the OpenType copy of Unifont.
It looks like it's a vectorized version of the bitmap font, with perfectly square pixels that meet at corners and so are non-manifold.
[image]
Again, applying a tiny offset works around the problem.

nophead, 2023-03-17:
Can you fix it with clipper by doing a tiny offset?

alidaf, 2023-03-27:
Try offset() in openSCAD with a very small number. I use this to fix some svg imports that would otherwise fail with the same error.

641i130, 2024-09-22:
Dealing with a similar issue. How should I go about fixing this?
```
/// This script will generate 3D objects of word tags for language learning purposes
scale([3,3,3]) {
    // word values
    word = "心配";
    word_h = 1;
    font_size = 4;
    // base values
    base = 4;
    height = 0.5;
    linear_extrude(height = word_h)
    translate([0, 0, 0])

    text(word, font="Noto Sans CJK JP", size = font_size, valign = "center", $fn = 100, halign= "center");
    // Make a rounded base
    difference() {
        hull() {
            translate([len(word)*2.75, 1, 0])
                cylinder(h=height, r = base, $fn = 100);
            translate([len(word)*-2.75, -1, 0])
                cylinder(h=height, r = base, $fn = 100);
            translate([-len(word)*2.75, 1, 0])
                cylinder(h=height, r = base, $fn = 100);
            translate([len(word)*2.75, -1, 0])
                cylinder(h=height, r = base, $fn = 100);
        }
    }
}
```

Some kanji pairs work, others do not. It renders fine in the view port, but not rendering with the render button.

kintel, 2024-09-22:
@jordanbrown0 Do you have a link to the font you could reproduce with?
If possible we should try to create a test case for this, but we need a font we can bundle with the sources.

In the meantime, could someone check if this works if Preferences->Features->manifold is enabled?

jordanbrown0, 2024-09-22:
I've changed computers since then and completely forgotten about this issue.

But a conversation with Mr. Google leads to
https://unifoundry.com/unifont/index.html
and indeed when I download and open
https://unifoundry.com/pub/unifont/unifont-16.0.01/font-builds/unifont-16.0.01.otf
I get an font sampler with an "install" button, and when I install it
```
text("o", font="Unifont");
```
yields the shape above.

That shape (with a cube, to force a real render) yields errors with CGAL but not with Manifold, on 2024.03.05.

@641i130 the workaround is, as discussed above, to apply a small offset before extruding.

kintel, 2024-09-22:
Thanks, good to hear that manifold handles this!

jordanbrown0, 2024-09-22:
Manifold seems to handle the cubes-touching-at-one-edge case, but I don't know whether that's intentional or accidental.  I'd check with the Manifold folks and ask what they expect.

kintel, 2024-09-22:
Manifold explicitly supports these use-cases. There are corner cases which are supported by OpenSCAD on a best-effort basis, but two distinct shapes touching in a vertex should always work.

jordanbrown0, 2024-09-22:
I believe CGAL handles the touching-vertex case.  It's the touching-edge case that's a problem for it.
```
cube(20);
translate([20,20]) cube(20);
```

kintel, 2024-09-22:
CGAL has partial support for this; it's representable as a Nef Polyhedron, but not as a regular Polyhedron, so we can evaluate it and export to STL, but we cannot re-import it.

jordanbrown0, 2024-09-22:
Ah.  I guess I was just scared away by the yellow WARNING with an exclamation point.

kintel, 2024-09-22:
There shouldn't be a yellow warning in manifold mode though. Did you get that?

jordanbrown0, 2024-09-22:
No, no warning for Manifold.  Just for CGAL.  (CGAL has scarred me for life.)

closed 2026-08-25 by github-actions[bot]