openscad/openscad #4559: F6 fails with fast-csg

jon-bondy, 2023-03-17:
Each time I run the program I run it from scratch, opening it and doing an F6 immediately.

If you run the attached program the following occurs:
- current 2021 release: succeeds and looks like this:

[image]

- lazy-union without fast-csg: succeeds
- fast-csg with or without lazy-union fails both in terms of error messages and failed geometry.

- fast-csg with lazy-union produces 
---
 
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
WARNING: Warning: mesh is not closed!
WARNING: [fast-csg] Corefinement corefinement mesh union #4 failed
ERROR: CGAL error in CGALUtils::applyUnion3DHybrid: CGAL ERROR: assertion violation!
Expr: e->incident_sface() != SFace_const_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h
Line: 329
WARNING: No top level geometry to render
---

- fast-csg without lazy-union produces
---
OpenSCAD 2023.02.23 
https://www.openscad.org/
 
Copyright (C) 2009-2022 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
 
Loaded design 'D:/Jon/Backdrop Holder.scad'.
 
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
WARNING: Warning: mesh is not closed!
WARNING: [fast-csg] Corefinement corefinement mesh union #4 failed
ERROR: CGAL error in CGALUtils::applyUnion3DHybrid: CGAL ERROR: assertion violation!
Expr: e->incident_sface() != SFace_const_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h
Line: 329
WARNING: Warning: mesh is not closed!
WARNING: [fast-csg] Corefinement corefinement mesh union #5 failed
ERROR: CGAL error in CGALUtils::applyUnion3DHybrid: CGAL ERROR: assertion violation!
Expr: e->incident_sface() != SFace_const_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h
Line: 329
WARNING: Warning: mesh is not closed!
WARNING: [fast-csg] Corefinement corefinement mesh union #9 failed
ERROR: CGAL error in CGALUtils::applyUnion3DHybrid: CGAL ERROR: assertion violation!
Expr: e->incident_sface() != SFace_const_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h
Line: 329
WARNING: Warning: mesh is not closed!
WARNING: [fast-csg] Corefinement corefinement mesh union #10 failed
ERROR: CGAL error in CGALUtils::applyUnion3DHybrid: CGAL ERROR: assertion violation!
Expr: e->incident_sface() != SFace_const_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h
Line: 329
WARNING: No top level geometry to render

Sometimes it actually produces a damaged image in the viewer:

[image]

- 

I cannot drop a SCAD file here so I renamed it to be a JPG file.  You will have to rename it back to a SCAD file.  Jeese.

[image]

jordanbrown0, 2023-03-17:
Here's the program.  (Sigh.  It's sad that they don't allow arbitrary files.)

```
// Backdrop Holder

inches = 25.4;
eps = 0.01;
$fn = 100;

id  = 0.709 * inches + 0.3;
idh = 0.709 * inches + 0.5;	//  horizontals are tighter
ir  = id /2;
irh = idh/2;
th = 2;
le = 1 * inches;

module Tube(le, di) {
	difference() {
		cylinder(h = le, d = di + 2*th);
		translate([0, 0, -eps]) cylinder(h = le + 2*eps, d = di);
		}
	}
	
module BottomTubes() 
	rotate([0, 0, -90]) {
		rotate([0, 90, 0])
			Tube(2*le, idh);
		rotate([0, -90, 0])
			Tube(2*le, idh);
		rotate([-90, 0, 0])
			Tube(3*le, idh);
		rotate([36.9, 0, 0]) rotate([-90, 0, 0])
			Tube(3*le, id);
		Tube(2*le, id);	
		}
module Bottom() {
	BottomTubes();
	translate([0, 0, -ir - th])
		linear_extrude(th)
			projection(cut = false)
				BottomTubes();
	}
!Bottom();
```

jordanbrown0, 2023-03-18:
Given that test program, I don't immediately duplicate the problem on 2023.02.23 on Windows.

kintel, 2024-11-28:
Closing as fast-csg has been deprecated

closed 2024-11-28 by kintel