openscad/openscad #4560: cannot be opened because the developer cannot be verified

MortenHofft, 2023-03-18:
I see this has been an issue in the past, but looks to have been resolved. I'm not able to open though.

dialog: `openscad cannot be opened because the developer cannot be verified` 2 buttons: `[Move to bin] | [Cancel]`

Going to system preferences and click "open anyway" doesn't help either. I just see the same message.

Chip: Apple M1 Pro
MacOS Ventura 13.2.1

[image]
[image]

I downloaded today 13 march 2023 [OpenSCAD-2023.03.13](https://files.openscad.org/snapshots/OpenSCAD-2023.03.13.dmg)
I also tried the 2021.01 version from the top on https://openscad.org/downloads.html

Both have the same issue. Are there any known solutions? Thanks

jmckeown2, 2023-05-27:
I just pulled 2023-05-26 to an Intel-based MBP running Ventura 13.4 and "Open Anyway" in Privacy Settings DID work as expected. From some Googling, it seem this is an M1 MacOS issue, not OpenSCAD. Internet peeps say something like
```
sudo xattr -d com.apple.quarantine /Applications/OpenSCAD.app
```
In terminal may be a fix. (make sure you use the correct path to the app.) Since I don't actually have the problem, I can't confirm if this helps to not.

gordio, 2023-07-21:
Just "Open Anyway" from System Settings — Privacy & Security **or** <right mouse click> on the app and choose Open and (optional: read message) again Open.
_It's 'coz Apple want $99/y for give you Developer cert._

gsohler, 2023-07-21:
> Just "Open Anyway" from System Settings — Privacy & Security *or* on the

I believe Same applies for Windows Version to Not Show big unverified blieb
Screen. But here ITS more espensive😕

philseeley, 2025-02-05:
What worked for me after opening the .dmg and dragging the .app into Applications, was to Right Click and select Open. This does something subtly different to double clicking and the warning dialog then gave me the option to run anyway.

gordio, 2025-02-05:
In macOS 15 and above you must explicitly approve it by opening the System Preferences app and navigating to the General tab in Security & Privacy.

kintel, 2025-02-05:
Closing this as fixed by #5420 
Note: The fix is only available in development snapshots, the last release (2021.01) is still unsigned due to unresolved signature issues, tracked in #5421

closed 2025-02-05 by kintel