openscad/openscad #4566: Manifold difference() across a gap leaves a stray face from the subtracted cube

wolfwood, 2023-03-19:
again evaluating @ochafik's heavy lifting on #4533 to resolve #391.

on this line https://github.com/wolfwood/tryadactyl/blob/try-1/testers/split-ortho-tester.scad#L28 a hollow part with tall walls is trimmed using a `difference()` with a large cube. in the result, with Manifold enabled, a stray face partially occludes the site of the difference.

[image]
[image]

let me know if there is other information I can provide.

pca006132, 2023-03-19:
@elalish I think we discussed about this and think that this is very hard to fix?
Wondering if we can make a pass to clean up features smaller than precision or something.

elalish, 2023-03-20:
Yes, I believe this is a precision issue, but we already do a pass to clean up features smaller than precision. Since we currently only use float instead of double precision, one must be a bit careful about using "large" cubes arbitrarily. If the cube is reduced to dimensions on the order of the rest of the object, does the problem persist? Manifold includes a `TrimByPlane` helper for this kind of case that automatically sizes the cube.

wolfwood, 2023-03-24:
the part is 110x80 and the cube is 500x500. how close is close enough?

I am somehow no longer able to reproduce this issue on either of two machines, when opening just this model. when I was encountered this I had opened a number of models and was switching back and forth between editor windows rendering. also toggling features and flushing caches. maybe it was an issue with context reuse?

elalish, 2023-03-24:
Interesting, hopefully so. The precision issue would be more likely if the cube's dimension is enough orders of magnitude (5 or 6?) bigger than the feature (wall) thickness to give floating-point trouble.

t-paul, 2023-04-13:
Seems to be fixed via Manifold update #4599.

Getting a warning about a missing include though.

[image]

```
Loaded design '/tmp/tryadactyl/testers/split-ortho-tester.scad'.
Parsing design (AST generation)...
[WARNING: Can't open include file '../../KeyV2/includes.scad'. in file ../key/cap.scad, line 8](https://github.com/openscad/openscad/issues/8,/tmp/tryadactyl/key/cap.scad)
Used file cache size: 7 files
Compiling design (CSG Tree generation)...
ECHO: "you need a 20 mm stem"
Rendering Polygon Mesh using Manifold...
Geometries in cache: 3328
Geometry cache size in bytes: 10820416
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:02.808
Top level object is a 3D object:
   Facets:       4366
Bounding box:
   Min:  -32.08, -48.54, -34.00
   Max:  84.08, 30.92, -7.47
   Size: 116.16, 79.46, 26.53
Rendering finished.
```

wolfwood, 2023-04-14:
thank you, the include is optional, not used by default. i wish i could do conditional includes. and that i could construct include file names using `str()`.

nophead, 2023-04-14:
The reason you can't is because includes are processed during syntax analysis, which is necessarily before variables are  given values.

pca006132, 2023-04-14:
if this is fixed by updating manifold, should this be closed?

t-paul, 2023-04-14:
@wolfwood could you please verify with the latest OpenSCAD builds that the issue is gone?

wolfwood, 2023-04-15:
@nophead I realized right after I said it that imports need to be processed to know the value of regular variables, lol.

@t-paul confirmed fix.

closed 2023-04-15 by wolfwood