openscad/openscad #4568: strings to numbers

germaniaherbert, 2023-03-20:
// Function "val" is missing 
// ========================
// in OpenSCAD 2019.05
// ===================
// There are numbers in a string Example:
st="00001204";
// with 
v=st[5];
// you get to the 5th value ->"2
echo(v); 
// this is now a string  
// now this value is to be used as index in another string
st2="12345678";
v2=st2[v];  
echo(v2);
// unfortunately this results in "undef"
// I can't find a function to convert to a
// numeric variable. In Basic I know "val"
// 
// thanks for help
 

Translated with www.DeepL.com/Translator (free version)

jordanbrown0, 2023-03-21:
Yes, a built-in string-to-number function would be good.  However, it's possible to write one in OpenSCAD-language.

BOSL2 has [one](https://github.com/revarbat/BOSL2/wiki/strings.scad#function-parse_int), for both integers and floating point.

Here's a simple one for integers if you want something standalone.

```
function int(s, ret=0, i=0) =
    i >= len(s)
    ? ret
    : int(s, ret*10 + ord(s[i]) - ord("0"), i+1);
```

But note that if you only need one digit, something like `ord(v)-ord("0")` will work.

germaniaherbert, 2023-03-21:
Hallo jordan,

vielen Dank für die schnelle Antwort. Obwohl ich in OpenSCAD schon viel programmiert habe, habe ich noch nie eine Funktion selbst entwickelt bzw. benötigt. Auch dies "ord(v)-ord("0")" habe ich getestet und es passt!!!
Nun kann ich unter Excel berechnete "Knoten", perfekt live darstellen.
(Animation). Die Werte und Farben kommen per 
include <datens.sca>;  
include <farben.sca>;

Beste Grüße und nochmals Dank sendet der germaniaherbert!

rcolyer, 2023-03-22:
I previously wrote a self-contained function for this, as a proof of concept (which people are free to use):  https://github.com/thehans/funcutils/blob/master/string.scad#L32-L53

That said, I completely concur that this belongs as a built-in feature of the language.  I wrote it because it was challenging, but it should not be, as the generalized notion of string to numerical value is a basic concept that deserves to be a built-in.

```
float = function(s) let(
    _f = function(s, i, x, vM, dM, ddM, m)
      i >= len(s) ? round(x*dM)/dM :
      let(
        d = ord(s[i])
      )
      (d == 32 && m == 0) || (d == 43 && (m == 0 || m == 2)) ?
        _f(s, i+1, x, vM, dM, ddM, m) :
      d == 45 && (m == 0 || m == 2) ?
        _f(s, i+1, x, vM, -dM, ddM, floor(m/2)+1) :
      d >= 48 && d <= 57 ?
        _f(s, i+1, x*vM + (d-48)/dM, vM, dM*ddM, ddM, floor(m/2)+1) :
      d == 46 && m<=1 ? _f(s, i+1, x, 1, 10*dM, 10, max(m, 1)) :
      (d == 69 || d == 101) && m==1 ? let(
          expon = _f(s, i+1, 0, 10, 1, 1, 2)
        )
        (is_undef(expon) ? undef : expon >= 0 ?
          (round(x*dM)*(10^expon/dM)) :
          (round(x*dM)/dM)/10^(-expon)) :
      undef
  )
  _f(s, 0, 0, 10, 1, 1, 0);
```

germaniaherbert, 2023-03-22:
Hallo rcolyer,

vielen Dank auch für Deine Antwort. Ich bin mal gespannt ob unser Anliegen Gehör findet und in einer neuen Version von OpenSCAD ein "val" Funktion fest eingebaut gibt.
Im übrigen habe ich nun gelernt wie man Funktionen codiert und mir selbst meine Funktion, (ich brauche nur ein Digit) entwickelt.

Beste Grüße vom germaniaherbert!

function val(d="") =
  // Umwandlung von String (eine Ziffer) zu numerisch
  // 20.03.2023 Herbert Schwarz
    len(d) != 1 ? undef :
    d[0] == "0" ? 0 :
    d[0] == "1" ? 1 :
    d[0] == "2" ? 2 :
    d[0] == "3" ? 3 :
    d[0] == "4" ? 4 :
    d[0] == "5" ? 5 :
    d[0] == "6" ? 6 :
    d[0] == "7" ? 7 :
    d[0] == "8" ? 8 :
    d[0] == "9" ? 9 :
    (0/0);

jordanbrown0, 2024-12-28:
English translation (from Google Translate):
> I previously wrote a standalone function for this, as a proof of concept (which people are free to use): https://github.com/thehans/funcutils/blob/master/string.scad#L32-L53
>
> That said, I completely agree that this belongs as a built-in feature of the language. I wrote it because it was challenging, but it shouldn't be, as the generalized notion of a string to a numeric value is a basic concept that deserves to be integrated.

germaniaherbert, 2024-12-28:
Hast du was neues geschrieben? Oder ist dies nun nur auf englisch? Grüße von Herbert Schwarz Von meinem Telefon gesendet-------- Ursprüngliche Nachricht --------Von: Jordan Brown ***@***.***>Datum: Sa., 28. Dez. 2024, 01:36An: openscad/openscad ***@***.***>Cc: germaniaherbert ***@***.***>, Author ***@***.***>Betreff: Re: [openscad/openscad] strings to numbers (Issue #4568)
English translation (from Google Translate):

I previously wrote a standalone function for this, as a proof of concept (which people are free to use): https://github.com/thehans/funcutils/blob/master/string.scad#L32-L53
That said, I completely agree that this belongs as a built-in feature of the language. I wrote it because it was challenging, but it shouldn't be, as the generalized notion of a string to a numeric value is a basic concept that deserves to be integrated.

jordanbrown0, 2024-12-28:
I don't think there's any English-only rule, but I believe the majority of the readers are English speakers so I thought I'd provide a translation so they didn't need to do it themselves.

The--Captain, 2025-10-03:
The solution posted here previously is lacking in that is does not account for multiple numbers in the string, and stumbles when presented with non-numerics mixed in.

Here's something I recently put together that grabs ALL the floats from a string.   The special case of [0-9]*.[0-9]*.[0-9]... is interpreted by simply ignoring subsequent decimal points until a non-numeric is reached.  For integers only it can easily be reduced, left as an exercise for the user.

Input is `my_string`, output is `num_list`

```
// String_To_Floats
// Jon Fox
// Creative Commons Non-Commercial Attribution License
my_string = "1,2 a2,2 .3,5dd 1.3234 5 -l2 -.64 .245 0.6 23. 663";  // yields  [1, 2, 2, 2, 0.3, 5, 1.3234, 5, 2, -0.64, 0.245, 0.6, 23, 663]
function chr_Ispart(S) = (chr_Isdec(S) || chr_Isnum(S));
function chr_Isdec(S) = (ord(S) == 46);
function chr_Isneg(S) = (ord(S) == 45);
function chr_Isnum(S) = ((ord(S) > 47) && (ord(S) < 58)); 
function num_Fromstr(S,ref_N,con_Val,exp_N) = (ref_N > -1) ? chr_Isnum(S[ref_N]) ? (exp_N == -1) ? num_Fromstr(S, ref_N - 1, con_Val + ((ord(S[ref_N])-48) * (10^0)), 0 + 1) : num_Fromstr(S, ref_N - 1, con_Val + ((ord(S[ref_N])-48) * (10^exp_N)), exp_N + 1) : chr_Isneg(S[ref_N]) ? con_Val*-1 : (exp_N == -1) ? con_Val : chr_Isdec(S[ref_N]) ? num_Fromstr(S, ref_N - 1, con_Val * (1/10^exp_N), -1 ) : con_Val : con_Val;
num_list = [ 
for(i = [0 : 1 : len(my_string)-1]) 
  if ((((my_string[i+1] == undef) || (!chr_Ispart(my_string[i+1]))) && chr_Ispart(my_string[i])) && !(chr_Isdec(my_string[i]) && !chr_Isnum(my_string[i-1])))
 num_Fromstr(my_string, i, 0, 0)
];
```

jordanbrown0, 2026-10-04:
I think that such a function should not attempt to do anything more than convert well-formed strings.  It should not attempt to process lists, or skip over "noise", or somehow handle extra decimal points; if the string is not valid it should return `undef`.  (I have mixed feelings on whether it should warn.)  At the most, it might strip leading and trailing whitespace.

Why?  Because silently "fixing" a user's "typos" requires guessing what the error is, and you have no reason to believe that you know what the error is.

If you find `3,500`, is the comma denoting a list (and so the result should be `[3, 500]`), or is it a thousands separator or noise (and so the result should be `3500`), or is it somebody accidentally typing a European-style number instead of an American-style number (and so the result should be `3.5`)?  If you find `1r3`, is the `r` noise (and so the result should be 13), or is it a position typo for `e` (and so the result should be `1000`)?  If you find `1.234.567`, is one of the decimal points an error (and so the result should be `1.234567` or `1234.567`) or is this again a European-style number with thousands separators (and so the result should be `1234567`)?

Silently doing something other than what the user intended is one of the worst things a program can do.

This is not to say that some of these features shouldn't be available, but they should be under the programmer's control.  There should be a "split string at delimiter" function, either built-in or user-written, to use if you expect a list.  (And such a function can correctly yield a one-entry list when there's no delimiter.)  "Strip out characters not in this set" is another possibility.

The--Captain, 2026-10-06:
To be frank my code was specifically written to be able to process openscads own output when printing a list of floats, although it has other uses. Modifying it to throw an error instead of continuing when it encounters non-numerics is trivial

jordanbrown0, 2026-10-07:
My comments are with respect to a hypothetical built-in feature. Your "user" program can do whatever works for you.