openscad/openscad #4574: Manifold - CGAL assertion failed with projection()

Virindi-AC2, 2023-03-23:
With Manifold enabled, the following is generated when "Design"->"Preview" is selected from the menu:

> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
> Expr: e_below != SHalfedge_handle()
> File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
> Line: 418

From the following code:

> list1 = [
> [115.57, 20.355],
> ];
>  
> list2 = [
> [-2.43, -2.43],
> [81.57, -2.43],
> ];
>  
> thickness = 10;
>  
> projection(cut=true)
> difference()
> {
>     translate([-24.46,-11.87,-thickness])
>     cube([264.06,138.7,thickness]);
>     for (iii=list1)
>     {
>     translate([iii[0],iii[1],-thickness-1])
>     cylinder(d=0.5,h=thickness+2,$fn=30);
>     }
>     for (iii=list2)
>     {
>     translate([140+iii[0],19.34+iii[1],-thickness-1])
>     cylinder(d=0.5,h=thickness+2,$fn=30);
>     }
> }

OpenSCAD: 2023.03.21.nightly
System: Debian Bullseye x86_64
Enabled features: manifold, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc
Advanced tab options:
[image]

Virindi-AC2, 2023-03-23:
Possibly another manifestation of the same problem? With the same settings, this similar code produces

> ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.

> list1 = [
> [-6.43, 4.48],
> [-6.43, 93.38],
> [-6.43, 109.255],
> [115.57, 109.255],
> [115.57, 20.355],
> [115.57, 4.48],
> ];
>  
> list2 = [
> [-2.43, -2.43],
> [81.57, -2.43],
> [-2.43, 41.57],
> [81.57, 41.57],
> ];
>  
> thickness = 19.23;
>  
> projection(cut=true)
> translate([0,0,1])
> difference()
> {
>     translate([-24.46,-11.87,-thickness])
>     cube([264.06,138.7,thickness]);
>     for (iii=list1)
>     union()
>     {
>     translate([iii[0],iii[1],-thickness-1])
>     cylinder(d=0.3,h=thickness+2,$fn=30);
>     translate([iii[0],iii[1],-thickness*2+5])
>     cylinder(d=0.3,h=thickness,$fn=30);
>     }
>     for (iii=list2)
>     union()
>     {
>     translate([140+iii[0],19.34+iii[1],-thickness-1])
>     cylinder(d=0.3,h=thickness+2,$fn=30);
>     translate([140+iii[0],19.34+iii[1],-thickness*2+5])
>     cylinder(d=0.3,h=thickness,$fn=30);
>     }
> }

gingerbeardman, 2023-10-03:
I'm also getting this: [.scad & .stl](https://github.com/openscad/openscad/files/12795371/Archive.zip)

```
WARNING: Warning: mesh is not closed!
WARNING: [fast-csg] Corefinement corefinement mesh intersection #0 failed with an error: CGAL ERROR: assertion violation!
Expr: it_poly_hedge!=edge_to_hedge.end()
File: /Users/distiller/libraries/install/include/CGAL/Polygon_mesh_processing/internal/Corefinement/Visitor.h
Line: 1528

ERROR: CGAL error in CGALUtils::applyOperator3DHybrid intersection: CGAL ERROR: assertion violation!
File: /Users/distiller/libraries/install/include/CGAL/Nef_3/SNC_const_decorator.h
Line: 276
Explanation: it is not possible to decide which one is a visible facet (if any)
WARNING: Warning: mesh is not closed!
WARNING: [fast-csg] Corefinement corefinement mesh difference #2 failed
ERROR: CGAL error in CGALUtils::applyOperator3DHybrid difference: CGAL ERROR: assertion violation!
File: /Users/distiller/libraries/install/include/CGAL/Nef_3/SNC_const_decorator.h
Line: 276
Explanation: it is not possible to decide which one is a visible facet (if any)
ERROR: Rendering cancelled by exception Unauthorized intersections of constraints
WARNING: No top level geometry to render```

pca006132, 2023-11-19:
works on latest master.

closed 2023-11-19 by kintel