openscad/openscad #4580: Mac image appears not to be properly PGP signed

christophberger, 2023-03-28:
I tried to verify the DMG image of OpenSCAD 2021.01 using the associated .asc file "OpenSCAD-2021.01.dmg.asc". However, GPG says it cannot find the signing key in my keychain.

GPG reports the signing key’s fingerprint as: E99C F73A 93E5 C42A D4E6  B8D8 0F30 F7BE D74A 1375

I fail to find a key with this fingerprint on keys.openpgp.org or other key servers.

The OpenSCAD download page lists this signing info:

- The OpenSCAD Developers [dev@openscad.org](mailto:dev@openscad.org)
- Key ID: [0x8AF822A975097442](http://keys.gnupg.net/pks/lookup?search=0x8af822a975097442&fingerprint=on&op=index)
- Fingerprint: B3C9 4B42 50DC 097E 9FFF 8177 8AF8 22A9 7509 7442

I cannot find any key with the above Key ID or fingerprint.

I can find a key for `dev@openscad.org` on `keys.openpgp.org`, but the fingerprint is different: E2EB DADD 336F F516 ADD5  1A78 F3E1 2CCC 2216 4A0F

So I have three different key fingerprints, an .asc file that I cannot use to verify the DMG file, and a key that supposedly belongs to the OpenSCAD developers but is not used anywhere.

Please help fix this issue, so that the Mac binary can be successfully verified.

kintel, 2023-03-31:
I'll look into it

esbenrug, 2024-02-11:
I tried to verify the windows installation file OpenSCAD-2021.01-x86-64-Installer.exe using OpenSCAD-2021.01-x86-32-Installer.exe.asc after having imported the openscad-signing-key_0x8AF822A975097442.asc

Doing that tells me that the key of "The OpenSCAD Developers <dev@openscad.org>" has expired

su-sd, 2025-02-09:
I just downloaded the latest snapshot release. It is signed with
```
E2EB DADD 336F F516 ADD5 1A78 F3E1 2CCC 2216 4A0F
E2EBDADD336FF516ADD51A78F3E12CCC22164A0F
```

The ([signing key](https://files.openscad.org/openscad-signing-key_0x8AF822A975097442.asc) in the download directory is also `0x8AF822A975097442`

closed 2026-08-25 by github-actions[bot]