openscad/openscad #4591: Text isn't centered if ttb or btt direction used

cymplecy, 2023-04-08:
and halign set to center

text("OpenSCAD",size=12,font="Liberation Mono",halign="center",valign="center",direction="ttb");

[image]

jordanbrown0, 2023-04-08:
Which version are you using?  I believe this is fixed in the development snapshot, in PR#3684.

[image]

jordanbrown0, 2023-04-08:
btt:

[image]

cymplecy, 2023-04-09:
I'm just using 2021.01 on Windows but glad to know it's been fixed in a dev version :)

closed 2023-04-09 by cymplecy