openscad/openscad #4595: OpenCSG rendering with edges: Z buffer tearing

kintel, 2023-04-11:
Originally discussed here: https://github.com/openscad/openscad/pull/1935#issuecomment-632873769

This issue has been root-caused as follows:

On some GPU hardware (at least recent Intel iGPU and nVidia), fixed function vertex transformations result in _very slightly_ different vertex positions than GLSL vertex transformations, even when GLSL is using the built-in client state uniform `gl_ModelViewProjectionMatrix`.

This, again, causes depth values produced by the fixed function pipeline to very slightly differ from depth values produced by GLSL vertex shaders.

OpenCSG currently uses fixed function vertex transformation (technically an ARB vertex program with ARB_position_invariant: https://github.com/floriankirsch/OpenCSG/blob/7b24d76556e885195c80952494454baf1544067c/src/renderSCS.cpp#L155), and OpenSCAD _usually_ also does fixed function. However, when rendering edges in preview mode, OpenSCAD uses a GLSL shader (https://github.com/openscad/openscad/blob/master/shaders/Preview.vert), which triggers z-buffer-tearing artifacts.

We have two tests which, when run on GPU hardware, fails due to this issue:
```
openscad-viewoptions-edges_view-options-tests
openscad-viewoptions-axes-scales-edges_view-options-tests
```

kintel, 2023-04-11:
Possible ways of addressing this:
* Quick-fix: Remove `ARB_position_invariant` from OpenCSG's shader and test if that produces values equal to GLSL.
* Future-proof fix: Rewrite OpenCSG's shaders to GLSL, rewrite OpenSCAD's renderer to always use GLSL shader in preview mode
* (Isolated fix in OpenSCAD: Rewrite wireframe renderer to use the fixed function pipeline. This is a regression towards legacy OpenGL and not desirable.)

kintel, 2023-05-28:
@jbinvnt This issue could be relevant to investigate. I expect similar issues to happen when using custom GLSL vertex shaders. At least this needs some careful testing.

damienmarchal, 2024-08-22:
I encountered this issue and confirm it when showing edge.

I suspect the issue may also impact the MouseSelector shader. As this is probably the shader used when right clicking on the 3d view this could explain why in some right click on the model seems fails... and slightly moving the mouse cursor makes it succeed.

damienmarchal, 2024-08-23:
I made a POC for a FIX. 

Here is the problematic rendering when using VBO+edge preview shader (probably similar issue with mouse picking shader) 
[image]

Here is the rendering with the FIX (still using VBO+edge preview shader). 
[image]

The fix consists in rendering the depth with the classical OpenCSG in a separated FBO and use it as texture for the color shader.  In the shader it is thus possible to implement a test that keeps pixels if the z-depths are nearly equal (and not stricly equals). Doing this also to cope with precision differences.

Note: 
the missing green on subtracted parts is my fault because I started from a custom implemented instead of master branch.

thehans, 2024-08-24:
It looks like you've also swapped the edge color with face color.  There is no longer any shading of faces based on normal.

damienmarchal, 2024-08-25:
@thehans you are right :) 

[image]

kintel, 2025-02-02:
@floriankirsch Thanks! This looks like a great solution, and will buy us some time. Eventually we might move off the compatibility profile (e.g. if we want to support GLES/WebGL), but we don't have any immediate plans for that.

damienmarchal, 2025-02-03:
@floriankirsch nice.

damienmarchal, 2025-02-04:
Hi all,

I give a quick to gl_Position = ftransform() in the ViewEdge shader, but it leads to no changes. 

 It is unclear to me if this may be connected to the compatibility profile or gl version.
Any idea ?

kintel, 2025-02-04:
One interesting detail could be that I believe OpenCSG isn't actually using the fixed function pipeline, but pre-glsl shaders. I'm not familiar with exactly how vertex processing is performed in fixed function vs. pre-glsl shaders vs. glsl shaders.

kintel, 2025-02-04:
But it doesn't explain why it works when we use the fixed pipeline. I slowly feel the need to write a minimal test app :/

damienmarchal, 2025-02-04:
I Forgot that part.

floriankirsch, 2025-02-04:
For the ARB shaders in OpenCSG, the respective option ARB_position_invariant is set as well. But maybe that option does not work correctly for specific drivers or hardware?!

Furthermore, ARB shaders in OpenCSG are only used for the SCS algorithm. The Goldfeather algorithm is using the fixed pipeline only. If you force the Goldfeather algorithm, does this resolve the artifacts.

kintel, 2025-03-12:
Fixed by #5725

kintel, 2025-03-12:
@damienmarchal I guess this would potentially unblock some of your picking work

closed 2025-03-12 by kintel

damienmarchal, 2025-03-12:
Hi @kintel 

This is good news, I will give a look this week.