openscad/openscad #4602: Icon preview

sonichy, 2023-04-15:
After install FlashPrint, stl files have icon preview in windows explorer.
Can OpenSCAD have icon preview in scad files ?
[image]

tpimh, 2023-06-16:
Yes, there's a script for thumbnailing .scad files: [OpenSCAD Thumbnailer](https://srlm.io/2015/12/15/scad-thumbnailer/).
Also one for .stl files: [OpenSCAD STL Thumbnailer](https://www.thingiverse.com/thing:258653). It can also be adapted to support other formats that OpenSCAD can import.

jordanbrown0, 2023-06-16:
Remember that it can take an arbitrary amount of time to preview an OpenSCAD file.  Usually it's just a few seconds, but it can be many seconds or longer.  A malicious program could run forever.

It isn't at all clear that it's a good idea to spend that time creating thumbnails by default.

And of course there are plenty of .SCAD files that are intended only for `include`, not for executing directly.

AJolly, 2024-12-24:
> Yes, there's a script for thumbnailing .scad files: [OpenSCAD Thumbnailer](https://srlm.io/2015/12/15/scad-thumbnailer/). Also one for .stl files: [OpenSCAD STL Thumbnailer](https://www.thingiverse.com/thing:258653). It can also be adapted to support other formats that OpenSCAD can import.
Note that thats for linux, not windows like the user originally asked.  I have yet to be able to find a way to generate thumbnails for openscad files on windows.  

And for time taken, thats what the OS's own thumbnail cache is for (and if you want, pregenerate the thumbnails)