openscad/openscad #4603: Preview is correct but rendering doesn't execute difference()

ptdecker, 2023-04-15:
Hello!

I love OpenSCAD but have encountered a problem that I am unable to figure out.  I am creating the edge ribs for a dodecahedron which will eventually be 3D printed.  The edge ribs will hold two-way mirror sheets and provide channels for LED strips and wiring behind the strips.  I created the 3D rib shape by extruding a 2D polygon that has the correct shape of the rib.  That all works fine.  Then I went to remove using difference() the equivalent of two miter cuts from one of the ends.  The final shape will need this on both ends, but doing it on a single end is okay for testing.  To do this, I created a 3D polyhedron that 'cuts' out the same material a miter saw cut would make. I then mirrored, rotated and translated a copy of this to take the cut on the other side of the rib.  I unioned() these two cuts and then use difference() to subtract this material from the extruded rib.

When I preview the rendering, it preview is fine and matches my goal.  But when I do a rendering it is as if the difference wasn't taken.  There are no errors on the console.

Any and all help is appreciated.

Here is the preview from two different perspectives:

[image]
[image]

Here is the render:

[image]
[image]

Here is the console output from the render:

```text
Parsing design (AST generation)...
Saved backup file: /Users/todddecker/Documents/OpenSCAD/backups/Edge Rib-backup-xJL33263.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 18
Geometry cache size in bytes: 120120
CGAL Polyhedrons in cache: 4
CGAL cache size in bytes: 206656
Total rendering time: 0:00:00.077
   Top level object is a 3D object:
   Simple:        yes
   Vertices:       38
   Halfedges:     114
   Edges:          57
   Halffacets:     42
   Facets:         21
   Volumes:         2
Rendering finished.
```

And, here is the code (sorry for its length):

```openscad
edge = 20;           // length of edge
edgeThickness = 1.5;  // thickness of the edge
ribWidth = 15;        // width of the rib
paneThickness = 3.25; // thickness of the pane
channelWidth = 11;    // width of LED strip channel
channelDepth = 1;     // depth of LED strip channel
wireWidth = 3 * 1.6;  // width of wire channel (3 22 AWG insulated wires)
wireDepth = 1.6;      // depth of wire channel (1 22 AWG insulated wire)
dihedral = 116.56505; // angle of polyhedron dihedral
interior = 108.0;     // angle of face interior
fudge = 0.01;         // render artifact elimination fudge factor

difference() {
    extruded_edge(edge, edgeThickness, ribWidth, paneThickness, channelWidth, channelDepth, wireWidth, wireDepth, dihedral);
    union() {
        angle_cut(ribWidth, interior);
        translate([ribWidth,0,0]) {
            rotate([0,0,180-dihedral]) {
                translate([ribWidth, 0, 0]) {
                    mirror([1,0,0]) {
                        angle_cut(ribWidth, interior);
                    }
                }
            }
        }
    }
}

module angle_cut(ribWidth, interior) {

    angle_ab = 90 - (interior / 2);

    ax = 0;
    ay = 0;
    az = 0;

    bx = ribWidth;
    by = 0;
    bz = 0;

    cx = 0;
    cy = 0;
    cz = ribWidth * tan(angle_ab);

    dx = 0;
    dy = 20;
    dz = 0;

    ex = ribWidth;
    ey = 20;
    ez = 0;

    fx = 0;
    fy = 20;
    fz = ribWidth * tan(angle_ab);

    polyhedron(
        points=[
            [ax, ay, az],  // point a (0)
            [bx, by, bz],  // point b (1)
            [cx, cy, cz],  // point c (2)
            [dx, dy, dz],  // point d (3)
            [ex, ey, ez],  // point e (4)
            [fx, fy, fz]   // point f (5)
        ],
        faces=[
            [0, 3, 5, 2], // face A
            [0, 1, 4, 3], // face B
            [1, 2, 5, 4], // face C
            [3, 4, 5],    // face D
            [0, 1, 2]     // face E
        ],
        convexity=10
    );
};

module extruded_edge(edge, edgeThickness, ribWidth, paneThickness, channelWidth, channelDepth, wireWidth, wireDepth, dihedral) {

    linear_extrude(height=edge, center=false, convexity=0, twist=0, slices=20, scale=1) {
        
        ax = 0;
        ay = 0;
        bx = ribWidth;
        by = 0;
        cx = ribWidth + ribWidth * cos(180 - dihedral);
        cy = ribWidth * sin(180 - dihedral);
        dx = cx - edgeThickness * cos(dihedral - 90);
        dy = cy + edgeThickness * sin(dihedral - 90);
        ex = dx - 0.667 * ribWidth * cos(180 - dihedral);
        ey = dy - 0.667 * ribWidth * sin(180 - dihedral);
        fx = ex - paneThickness * cos(dihedral - 90);
        fy = ey + paneThickness * sin(dihedral - 90);
        gx = fx + 0.667 * ribWidth * cos(180 - dihedral);
        gy = fy + 0.667 * ribWidth * sin(180 - dihedral);
        px = 0;
        py = edgeThickness + paneThickness;
        qx = 0.667 * ribWidth;
        qy = py;
        rx = qx;
        ry = edgeThickness;
        sx = 0;
        sy = ry; 

        angle_opq = atan((gy - py) / gx);  // derived angle of inside edge
        gp = gx / cos(angle_opq);          // total width of inside edge
        gh = 0.5 * (gp - channelWidth);    // width of offset on either side of the LED channel
        ho = gp - 2 * gh;                  // derived width of LED channel (should match specified)
        jm = wireWidth;                    // total width of wire channel
        ij = 0.5 * (ho - jm);              // width of offset on either side of wire channel within the LED channel
        assert(ho == channelWidth, "Derived channel width does not match specified width");
        assert(jm < ho, "Total wire channel width exceeds the width of the LED strip");

        hx = gx - gh * cos(angle_opq);
        hy = gy - gh * sin(angle_opq);
        ix = hx + channelDepth * sin(90 - 2 * angle_opq);
        iy = hy - channelDepth * cos(90 - 2 * angle_opq);

        ox = px + gh * cos(angle_opq);
        oy = py + gh * sin(angle_opq);
        nx = ox + channelDepth * sin(90 - 2 * angle_opq);
        ny = oy - channelDepth * cos(90 - 2 * angle_opq);
        
        jx = ix - ij * cos(angle_opq);
        jy = iy - ij * sin(angle_opq);

        kx = jx + wireDepth * sin(90 - 2 * angle_opq);
        ky = jy - wireDepth * cos(90 - 2 * angle_opq);

        mx = nx + ij * cos(angle_opq);
        my = ny + ij * sin(angle_opq);
        lx = mx + wireDepth * sin(90 - 2 * angle_opq);
        ly = my - wireDepth * cos(90 - 2 * angle_opq);
        

        
        polygon([
            [ax, ay], // point 'a'
            [bx, by], // point 'b'
            [cx, cy], // point 'c'
            [dx, dy], // point 'd'
            [ex, ey], // point 'e'
            [fx, fy], // point 'f'
            [gx, gy], // point 'g'
            [hx, hy], // point 'h'
            [ix, iy], // point 'i'
            [jx, jy], // point 'j'
            [kx, ky], // point 'k'
            [lx, ly], // point 'l'
            [mx, my], // point 'm'
            [nx, ny], // point 'n'
            [ox, oy], // point 'o'
            [px, py], // point 'p'
            [qx, qy], // point 'q'
            [rx, ry], // point 'r'
            [sx, sy]  // point 's'
        ]);
    }
};
```

ptdecker, 2023-04-16:
I noticed that when I try to render the very first time, I do receive the following error.  This error only appears once.  If I 'preview' then 'render' again, the error no longer occurs in the log:

```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
**ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. 
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.** 
Geometries in cache: 8
Geometry cache size in bytes: 61344
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 51664
Total rendering time: 0:00:00.202
   Top level object is a 3D object:
   Simple:        yes
   Vertices:       38
   Halfedges:     114
   Edges:          57
   Halffacets:     42
   Facets:         21
   Volumes:         2
Rendering finished.
```

ptdecker, 2023-04-16:
There is a related problem with the same error here:

https://github.com/openscad/openscad/issues/4039

jordanbrown0, 2023-04-17:
Yes, and the thread at #4039 has a number of good hints.

The first thing to learn is that OpenSCAD caches the results of renders, including the results of *failing* renders.  That cache can mean that your next F6 does not attempt to render the troublesome subassembly.  When you're trying to diagnose this kind of problem, use Design / Flush Caches before each F6 attempt.

The next thing to learn is about debugging polyhedra.  Read the [polyhedron section](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron ) of the manual and particularly the "Debugging polyhedra" section.  In this particular case, you have one side that's built backwards; it lists the faces counter-clockwise (as viewed from the outside of the polyhedron).  In this case it's made more difficult to see because it's one of the "negative" angle_cut() polyhedra that is subtracted from the main assembly.  If you isolate that one polyhedron with the "!" modifier and turn on View / Thrown Together, you'll see that one side is purple:
[image]
This is the face that you call face E; changing it from [0,1,2] to [0,2,1] fixes the problem.

Next thing:  see the transparent parts of your model when you preview?  That's a convexity problem.  All of the operations that build 3D objects *other* than the primitives have a "convexity" parameter.  Theoretically you should set it to the number of times that a line crossing the object might transition into and out of the object.  Practically, setting it to 10 or so is usually enough and rarely has any perceptible performance impact.  Ooh, and I see that you set convexity=0.  I don't know whether zero means the same as one, but for this object it looks like the theoretical value is around four.  Just set it to ten.  See also [this frequently asked question](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?).

Next:  see the flickering areas where stuff has been cut away?  That's called Z-fighting.  The previewer hates it when the object that you are subtracting has a face in the same place as the thing it's being subtracted from.  See [this frequently asked question](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?).  The solution is to have the negative object extend further outside the object it's being subtracted from.  Some people use small fractions and give them a name like "epsilon"; I usually just add one appropriately, since it's the only reason that I would add one. The first step is to subtract one from the Y coordinates in the first three points in the polyhedron so that that face moves to Y=-1.  For the other two faces, it's a little trickier because you want to keep the corner point at the same place while moving a side.  For this particular shape, you could move the corner that's not being used for the cut down and to the left.  Subtract one from ax and dx, and one from az and dz.

Here's the new polyhedron call, reversing that face and making it a little bigger in the right areas to fix the Z-fighting:
```
    polyhedron(
        points=[
            [ax-1, ay-1, az-1],  // point a (0)
            [bx, by-1, bz],  // point b (1)
            [cx, cy-1, cz],  // point c (2)
            [dx-1, dy, dz-1],  // point d (3)
            [ex, ey, ez],  // point e (4)
            [fx, fy, fz]   // point f (5)
        ],
        faces=[
            [0, 3, 5, 2], // face A
            [0, 1, 4, 3], // face B
            [1, 2, 5, 4], // face C
            [3, 4, 5],    // face D
            [0, 2, 1]     // face E
        ],
        convexity=10
    );
};
```
Note:  You do *not* need convexity here, because this polyhedron is entirely convex.

Another thing that you might to do simplify:  you don't really need a custom polyhedron there.  There's no problem with having your negative object be larger than your positive object.  Your cut is a simple plane cut.  You could just use a pair of cubes that are appropriately rotated so that their top faces are in the right places, and that are bigger than the section they need to cut.  The exact math is left as an exercise for the reader, but I suspect that it involves a rotate by `angle_ab`.

Finally, yeah, the example is kind of long.  Knowing that you need to flush caches should make it easier to diagnose, and using the "!" and "#" modifiers can help a lot to understand what's going on with each individual component.

jordanbrown0, 2023-04-17:
My pleasure.  [Pay it forward.](https://en.wikipedia.org/wiki/Pay_it_forward)

jordanbrown0, 2023-04-17:
While there are a number of issues discussed here where it it would sure be nice to improve OpenSCAD's behavior (pickiness about polyhedron definition, Z-fighting, need to set convexity), those are longstanding and well-understood limitations.

The fact that render results are cached even for failing renders is discussed in #454.

Net, there's nothing left to talk about in this issue.  Would you please be so good as to close it?

ptdecker, 2023-04-17:
Closing.  Thank you for the wonderful support

closed 2023-04-17 by ptdecker