openscad/openscad #4605: Feature request DXF: Circle output as CIRCLE instead of POLYLINE

ukd1, 2023-04-15:
Whilst I mostly 3D print and make things in OpenSCAD (and love it), if I want anything CNC / Laser cut I generally have to switch to using another CAD program (e.g. cadquery, or Fusion360*) as OpenSCAD doesn't output Circle as `CIRCLE` in DXF. Outputting as a `CIRCLE` is "required" for sendcutsend, especially if you want to get them to cut threaded holes, but they usually complain anyway.

This is a message from them:

```
Unfortunately OpenSCAD is great for parametric design, but it doesn't work great for generating tool paths due to the polygon approximation.
```

They then suggested using QCAD. I've since found cadquery outputs `CIRCLE` as expect, but I'd rather not have to relearn.

For a sample:

```openscad
// adding this does not solve the issue, just makes it worse by having more edges to cut. removing it, still has the issue but default number of edges.
$fn = 1000; 

difference(){
    circle(r=50);
    circle(r=8.21);
}
```

Is this something OpenSCAD would consider adding / changing, or even merging if there was a PR for it? or does this break other things, and it'll never change?

* p.s. the changing / relearning is a pain for me, as this is a hobby and I have limited time to learn / be good at it

ukd1, 2023-04-16:
If there is no chance of fix / changes....I'm also interested if there any workarounds for this, I'd also love to know; either export format changes, code changes, or even a vendor change if anyone knows of one that can somehow do this automatically?

jordanbrown0, 2023-04-16:
Alas, turning all 2D objects into sets of line segments, and all 3D objects into sets of faces, is pretty fundamental to OpenSCAD operation.

It seems plausible (though perhaps difficult) that a standalone circle might be kept as a circle further in the pipeline, but definitely as soon as you do any union/intersection/difference that touches it, it'd have to turn into line segments.

nophead, 2023-04-16:
For Mendel90 I converted DXF files into SVG files with true circles in them. I did that to make drill templates with a cross on the centre and circle with a labelled diameter. The Python code is here: https://github.com/nophead/Mendel90/blob/master/dxf.py

It recognises polygons that are regular and have more than 8 sides. The same technique could be used to output a new DXF with circle elements in it.

t-paul, 2023-04-16:
> Is this something OpenSCAD would consider adding / changing

Yes

> or even merging if there was a PR for it?

Yes, considering this may not have a high priority for most people, that would hugely increase chances.

However, this will need a solution that is generally usable and not a "simple" workaround for some specific cases. This does not mean it would need to actually implement everything at once, small steps are actually prefered. But the way of approaching this topic would need to be applicable for other things too, ideally also 3D at some point. This very much connects to the feature request of exporting STEP files with internal structure as opposed to just a single mesh.

So this may be quite a bit more complex than obvious at the first glance, but would be a useful feature.

pitch-circle, 2023-04-17:
...I'm also interested if there any workarounds for this...

Download FreeCAD. Open .SCAD file therein. Export as dxf. Minimal extra learning. This process could be automated.

ukd1, 2023-04-17:
@pitch-circle, tbh, I downloaded freecad to redo a project and found it super steep (vertical) learning curve for me. My current work around was to use cadquery, which just worked (after rewriting my code...)......... so just a simple import/export moves POLYLINES --> CIRCLE?

ukd1, 2023-04-17:
> For Mendel90 I converted DXF files into SVG files with true circles in them. I did that to make drill templates with a cross on the centre and circle with a labelled diameter. The Python code is [...]

Interesting will check this out!

jordanbrown0, 2023-04-17:
Any solution to this problem should consider cases where the circle is combined with another object, e.g.:
```
circle(10);
square(10);
```
Carrying **arcs** through seems like a huge win.  Whether detecting and repairing pure circles is valuable... don't know.

Note that I say "consider"... that is, if the solution only covers pure circles, that's OK, just be clear about the limitation.

ukd1, 2023-04-17:
@jordanbrown0 that would be valuable, though my (and I presume is more common) use case is mostly for holes, which at least is simpler...I presume.

Even though I'm a software engineer and would love to attempt a fix myself, I'm not really sure where I'd be starting on this. The external script to "fix" it seems like a hack, and is kinda annoying workflow wise, but I've not done c++ for 20 years now, lol.

--> IRC!

pca006132, 2023-04-18:
I think some CAD programs and slicers such as prusaslicer can recover features such as arcs and faces from a mesh. Maybe OpenSCAD can provide some options to specify error threshold and recover those features before DXF export?

Here is the issue in prusaslicer: https://github.com/prusa3d/PrusaSlicer/issues/3651
And this seems to be their implementation: https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/Measure.cpp

pca006132, 2023-04-18:
True. I am not familiar with the internals of openscad, but for operations supported by manifold, you can attach properties to each vertex, so perhaps each primitive call can create a new id, and when performing exports you can map the ids back to primitive features. (not the best solution, just something I can immediately think of)

t-paul, 2023-04-18:
Oh, nice. Something like this is already used for the "jump to source" feature in preview mode. This is worth exploring. The other options I see is mostly keeping the whole tree available so the export does not just get handed the final mesh, but instead the geometry tree.

The "stupid" export could just force mesh calculation and go on as before, but it could also go trough the tree an try generating the export based on the still available primitives only generating meshes where it's not possible to handle otherwise.

ukd1, 2023-04-21:
@elalish for me, it's getting `CIRCLE` in `DXF` output; it's required by the company (sendcutsend.com) that I use for cutting designs in to metal / plastic.