openscad/openscad #4606: mass check scad file for errors

TylonHH, 2023-04-16:
Hey guys,

how can I validate a file? Imagine a have 1.000 scad files. I wanna check if they can generate a model. If not I need the debugging the message. Is this maybe possible with API? Because it will be done not by human, it would be awesome if its possible with non visual interface.

cheers

gsohler, 2023-04-16:
you can use openscad in command line mode with -o <out.png> to see if out.png creates a picture which is not empty.

t-paul, 2023-04-16:
See also [Using OpenSCAD in a command line environment](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment)

Additionally, there are also tools for static code analysis like:
https://gitlab.com/bath_open_instrumentation_group/sca2d

For further general questions, please use the [mailing list](https://openscad.org/community.html#forum).

closed 2023-04-16 by t-paul