openscad/openscad #4612: Majority of tests require -DEXPERIMENTAL=ON

kintel, 2023-04-18:
Currently, ~88% of tests fail if we forget to add EXPERIMENTAL=ON when configuring OpenSCAD.
While this is fine for development of experimental features, it leaves the test coverage for the OpenSCAD release build in a pretty weak state.

We should consider having great test coverage for all non-experimental features and run these by default.
Any extra tests added for experimental builds could be run in addition (or added to a separate test profile), when experimental features are enabled.

thehans, 2023-04-19:
Some more info for context and discussion.  

About the failure mode:  most of these tests are actually failing to run OpenSCAD at all, because it doesnt' recognize the command line `--enable` params and just spits out command line usage help text.  So no "actual" image output is created and the python image comparison rightly fails.

In the context of tests and experimental features, I'd say test cases are divisible into two notable categories:
1) Tests cases which are specific to an experimental feature.  If a test was written specifically to test a given feature, and does not make sense to run without said feature.
   - If `EXPERIMENTAL=OFF`, we should make sure that cmake sets these tests to disabled.
   - If `EXPERIMENTAL=ON` then include the test case in the test suite run as usual.
2) Tests cases which were not written specifically for a experimental feature, BUT we may still like to verify that it passes when that feature is enabled.
   - If `EXPERIMENTAL=OFF` then run the test with no "--enable" arguments
   - If `EXPERIMENTAL=ON` then probably we should try to run the test twice: first with **NO** "--enable" arguments, **THEN** *with them*.

One further complication, and I think at least partial explanation of why things ended up this way, is that the old CGAL Nef code colors faces a greenish tone if they result from a `difference` or `intersection` operation.  while fast-csg only uses the main cornfield yellow.  So, many of the CGAL expected images had to be updated to match the yellow-only output of fast-csg, in order for fast-csg experimental tests to pass.  

Therefore I think we will need to disable that CGAL code which colors cut faces differently before we can move forward with this issue.
...wherever that code is... I've briefly searched a couple times in the past  but never quite found where/how that gets set.

kintel, 2023-04-19:
Thanks for the writeup!

In terms of colors; we can also run CGAL stuff with a custom color scheme, now that all golden images are updated. I _believe_ that's possible.