openscad/openscad #4613: Support headless EGL context creation using the standard binary

kintel, 2023-04-19:
This is a follow-up to #3857:

We currently have EGL support in OpenSCAD, which allows us to create OpenGL (software or hardware) contexts in headless mode. However, since we depend on GLEW and GLEW is configured using either GLX or EGL at compile-time, we have to create a separate binary for this feature: https://github.com/openscad/docker-openscad/blob/main/openscad/bookworm/Dockerfile.egl. This separate binary also have to contain custom built GLEW and OpenCSG libraries.

Without EGL, we have to create some X11 display in headless mode, typically using Xvfb or Xvnc. This is also how the OpenSCAD test framework works today.

**Feature Request**
* Allow using the standard OpenSCAD binary to request an offscreen context in headless mode

Various notes/followups:
* We most probably need to get rid of GLEW.
* OpenCSG still uses GLEW; that may have to change.
* Should we offer a cmd-line option for choosing GLX vs. EGL or can we be robust enough to make that decision automatically?
* Test framework should be updated once this lands
* We may need to offer some docs for getting people past Linux rendering node permission settings

praetp, 2024-06-03:
Any update on this ?

t-paul, 2024-06-03:
I believe that issue is already solved when building with GLAD / Included-OpenCSG (like the `dev` Docker Images). @kintel?

kintel, 2024-06-04:
Yes, this can be closed. When building against OpenCSG >= 1.6, we automatically enable EGL mode, so Linux binaries will get this feature as long as the newer OpenCSG gets picked up. For distros shipping with an older version, `cmake -DUSE_BUILTIN_OPENCSG=1` can be used. Our Linux tests do this and no longer need to deal with virtual X11 displays:
https://github.com/openscad/openscad/blob/master/scripts/github-ci.sh#L35

closed 2024-06-04 by kintel