openscad/openscad #4614: Flaky test: manifold-cgalpng_example006

kintel, 2023-04-19:
The test `manifold-cgalpng_example006` is flaky on the Linux CI, both on Ubuntu-20.04 and Ubuntu-22.04:
```
run_test() cmdline: /home/runner/work/openscad/openscad/b/openscad /home/runner/work/openscad/openscad/examples/Old/example006.scad --enable=vertex-object-renderers --enable=manifold --render -o /home/runner/work/openscad/openscad/b/tests/output/manifold-cgalpng/example006-actual.png
using font directory: /home/runner/work/openscad/openscad/tests/data/ttf
stderr output: ECHO: version = [2023, 4, 19]

Error: openscad failed with return code -11
```

Return code -11 is most likely a segfault: For return from a segfault, the error code on Linux is 139. That's bit 7 being 1 to indicate a signal, and 11 in the lower 7 bits to indicate signal 11, segfault. ctest is then reporting this as "-11".

Observing this over a month or so gives me the impression that it fails in somewhere around 5-20% of the runs, but always with the same failure. Data mining the CI logs may give a better answer.

kintel, 2023-04-19:
I ran this test 1000x locally with no failures, so the CI environment may contribute heavily towards this.

pca006132, 2023-04-20:
Maybe you can try to compile it with debug information and address sanitizer enabled? We found it very useful in finding segfault related bugs.

kintel, 2023-05-04:
Building OpenSCAD with thread sanitizer, this command-line produces 13 MB worth of data races on Ubuntu: `./openscad ../examples/Old/example001.scad --enable=manifold -o out.stl`.

It feels a bit much, and since all dependencies (including tbb) was installed with apt-get and thus built without sanitizing, they could be largely false positives.

Some examples:
```
WARNING: ThreadSanitizer: data race (pid=191408)
  Read of size 8 at 0x7fdecbb13890 by thread T7:
    #0 bool tbb::detail::d1::dynamic_grainsize_mode<tbb::detail::d1::adaptive_mode<tbb::detail::d1::auto_partition_type> >::check_being_stolen<tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<long>, thrust::system::tbb::detail::for_each_detail::body<thrust::zip_iterator<thrust::tuple<thrust::zip_iterator<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>*, int*, int*, int*, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> >, thrust::pointer<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::system::tbb::detail::par_t, thrust::use_default, thrust::use_default>, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> >, long, thrust::detail::allocator_traits_detail::copy_construct_with_allocator<thrust::detail::no_throw_allocator<thrust::detail::temporary_allocator<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::system::tbb::detail::par_t> >, thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> > >, tbb::detail::d1::auto_partitioner const> >(tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<long>, thrust::system::tbb::detail::for_each_detail::body<thrust::zip_iterator<thrust::tuple<thrust::zip_iterator<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>*, int*, int*, int*, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> >, thrust::pointer<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::system::tbb::detail::par_t, thrust::use_default, thrust::use_default>, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> >, long, thrust::detail::allocator_traits_detail::copy_construct_with_allocator<thrust::detail::no_throw_allocator<thrust::detail::temporary_allocator<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::system::tbb::detail::par_t> >, thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> > >, tbb::detail::d1::auto_partitioner const>&, tbb::detail::d1::execution_data const&) /usr/include/oneapi/tbb/partitioner.h:421 (openscad+0x127498d)
    #1 tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<long>, thrust::system::tbb::detail::for_each_detail::body<thrust::zip_iterator<thrust::tuple<thrust::zip_iterator<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>*, int*, int*, int*, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> >, thrust::pointer<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::system::tbb::detail::par_t, thrust::use_default, thrust::use_default>, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> >, long, thrust::detail::allocator_traits_detail::copy_construct_with_allocator<thrust::detail::no_throw_allocator<thrust::detail::temporary_allocator<thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::system::tbb::detail::par_t> >, thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type>, thrust::tuple<glm::vec<4, float, (glm::qualifier)0>, int, int, int, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type, thrust::null_type> > >, tbb::detail::d1::auto_partitioner const>::execute(tbb::detail::d1::execution_data&) /usr/includ [... cut ...]
    #8 GeometryEvaluator::evaluateGeometry(AbstractNode const&, bool) /home/kintel/code/OpenSCAD/openscad/src/geometry/GeometryEvaluator.cc:72 (openscad+0x81d879)
    #9 do_export(CommandLine const&, RenderVariables const&, FileFormat, SourceFile*) /home/kintel/code/OpenSCAD/openscad/src/openscad.cc:543 (openscad+0x2e7dfc)
    #10 cmdline(CommandLine const&) /home/kintel/code/OpenSCAD/openscad/src/openscad.cc:426 (openscad+0x2eb061)
    #11 main /home/kintel/code/OpenSCAD/openscad/src/openscad.cc:1189 (openscad+0x166e28)

```

pca006132, 2023-05-12:
We just fixed a bug in https://github.com/elalish/manifold/pull/430, wondering if it can be related as it will cause segfault, although it should be pretty deterministic...

For the race condition, it seems that they are due to thrust, and it is known that they sometimes have UB in their code...

pca006132, 2023-12-02:
is this still flaky?

closed 2023-12-03 by kintel