openscad/openscad #4617: Non-determinism with manifold and minkowski

thehans, 2023-04-22:
**Describe the bug**
I was testing a scad file, which makes heavy use of minkowski, on the new manifold feature in nightlies.

Every render is giving different results.  
  - Sometimes it successfully outputs the geometry
     - Although, the number of facets varies for every success
  - Sometimes it hits a CGAL assertion (there are at least 2 different possible assertions I saw)
  - Sometimes it hangs indefinitely (I observed one run taking 30+ minutes, manually killed)
  - At least once I observed it memory leak until 32GB RAM + 32GB swap was used and then OOM killed.

**To Reproduce**

Steps to reproduce the behavior:
1. Render the scad script provided below. `openscad-nightly --enable manifold ./test.scad -o out.stl`
2. Repeat
3. Observe different results each time.

Log from terminal:
```
$ openscad-nightly --enable manifold ./test.scad -o out.stl
Geometries in cache: 51
Geometry cache size in bytes: 104200                                                                                                                                                           
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0         
Total rendering time: 0:00:12.873
Top level object is a 3D object:
   Facets:      10276            
$ openscad-nightly --enable manifold ./test.scad -o out.stl
Geometries in cache: 51
Geometry cache size in bytes: 104200                                                                                                                                                           
CGAL Polyhedrons in cache: 0                                                                                                                                                                   
CGAL cache size in bytes: 0
Total rendering time: 0:00:12.748                                                              
Top level object is a 3D object:
   Facets:       9320
$ openscad-nightly --enable manifold ./test.scad -o out.stl
Geometries in cache: 51                    
Geometry cache size in bytes: 104200
CGAL Polyhedrons in cache: 0                                                                   
CGAL cache size in bytes: 0
Total rendering time: 0:00:12.989
Top level object is a 3D object:                                                                                                                                                               
   Facets:       9932  
$ openscad-nightly --enable manifold ./test.scad -o out.stl
Geometries in cache: 51     
Geometry cache size in bytes: 104200
CGAL Polyhedrons in cache: 0     
CGAL cache size in bytes: 0     
Total rendering time: 0:00:13.417
Top level object is a 3D object:                                                                                                                                                               
   Facets:       9548                                                                                                                                                                          
$ openscad-nightly --enable manifold ./test.scad -o out.stl
Geometries in cache: 51                                                                        
Geometry cache size in bytes: 104200
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0                                                                    
Total rendering time: 0:00:12.681          
Top level object is a 3D object:
   Facets:      10056                                                                          
$ openscad-nightly --enable manifold ./test.scad -o out.stl
Geometries in cache: 51
Geometry cache size in bytes: 104200                                                           
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:13.121
Top level object is a 3D object:
   Facets:       9958
130 $ openscad-nightly --enable manifold ./test.scad -o out.stl
Geometries in cache: 51
Geometry cache size in bytes: 104200
[... 680 lines cut ...]
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
```  
</details>

**Additional context**
The script is meant to experiment with [morphology operations](https://en.wikipedia.org/wiki/Mathematical_morphology) and composing them into a generic "roundover" operation which can fillet both convex and concave edges, without affecting the bounding box of the object.

In 2D, we can use offset and these are trivial to implement.  But in 3D the closest thing we have to offset is minkowski with a sphere.  
Since `minkowski` *addition* can't directly simulate a *negative* offset, a number of additional operations are done to act as  geometric complement, operating on that with positive offsets, etc.

I was originally testing with an octahedral geodesic sphere from a library of mine, but replaced with standard sphere so there would be no external dependencies.  Near the top of the script the `use` directives can be un-commented and the commented line inside `sphere_for_offset` can be swapped if interested in trying.  The same types of issues result either way.

thehans, 2023-04-23:
ping @ochafik

pca006132, 2023-11-19:
If you enable manifold deterministic option:

```diff
diff --git a/src/openscad.cc b/src/openscad.cc
index f4646bbf3..8ed411d8a 100644
--- a/src/openscad.cc
+++ b/src/openscad.cc
@@ -51,6 +51,10 @@
 #include <vector>
 #include <fstream>
 
+#ifdef ENABLE_MANIFOLD
+#include "manifold.h"
+#endif
+
 #ifdef ENABLE_CGAL
 
 #include "CGAL_Nef_polyhedron.h"
@@ -918,6 +922,10 @@ int main(int argc, char **argv)
   init_mimalloc();
 #endif
 
+#if ENABLE_MANIFOLD
+  manifold::ManifoldParams().deterministic = true;
+#endif
+
   int rc = 0;
   StackCheck::inst();
 
```

and disable parallel minkowski by setting `OPENSCAD_NO_PARALLEL=1`, you will have deterministic output.

```
Geometries in cache: 10
Geometry cache size in bytes: 293984
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:05.066
Top level object is a 3D object:
   Facets:       8854
```

Btw, interestingly, parallel minkowski is slower than the sequential one.

pca006132, 2025-03-17:
This should be fixed after the latest manifold update.

closed 2025-03-17 by pca006132