openscad/openscad #4623: [manifold] Minkowski failed with error, falling back to Nef operation: CGAL ERROR: assertion violation!

Loki44, 2023-05-02:
**Describe the bug**
Rendering the code below produces the following error when the "manifold" option is turned on, it renders fine when it's off:
```
WARNING: [manifold] Minkowski failed with error, falling back to Nef operation: CGAL ERROR: assertion violation!
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Convex_decomposition_3/SM_walls.h
Line: 438
Explanation: wrong handle

ERROR: Rendering cancelled by exception CGAL ERROR: assertion violation!
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Convex_decomposition_3/SM_walls.h
Line: 438
Explanation: wrong handle
WARNING: No top level geometry to render
```

**To Reproduce**
Steps to reproduce the behavior:
1. enable manifold option
2. open code
3. click render
4. wait
5. See error

**Expected behavior**
it renders without error

**Code reproducing the issue**
It's also in the attached .txt file
```
tiny = 0.001;

$fa = $preview ? 12 : 1;
$fs = $preview ? 2 : 0.5;

module chamfer_text(words,text_width,text_height,font){
    difference() {
        linear_extrude(height = text_height) {
            resize([text_width,0,0], auto=true) {
                mirror([1,0,0]){
                    text(text = words, valign = "center", halign="center",font = font);
                }
            }
        }

        minkowski() {
            difference() {
                translate([0,0,text_width+tiny]) {
                    cube(text_width*2,center=true);
                }

                linear_extrude(height = text_height) {
                    resize([text_width,0,0], auto=true) {
                        mirror([1,0,0]){
                            text(text = words, valign = "center", halign="center",font = font);
                        }
                    }
                }
            }

            cylinder(h = text_height , r1 = tiny, r2 = 1);
        }
    }
}

chamfer_text(words = "test", text_width = 50, text_height = 1.5, font = "Liberation Sans");
```

**Environment and Version info (please complete the following information):**
 - OS: windows 10
 - System: Intel PC 64-bit
 - OpenSCAD Version 2023.04.25 (git bb517c3bf), from the developer snapshots on the website

**Library & Graphics card information**
```
OpenSCAD Version: 2023.04.25 (git bb517c3bf)
System information: Microsoft Windows 10 (10.0.19045) x86_64 8 CPUs 7.88 GB RAM
User Agent: OpenSCAD/2023.04.25 (git bb517c3bf) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "11.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_80
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.8
QScintilla version: 2.11.2
InputDrivers:
GLib version: 2.70.2
lodepng version: 20210627
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.0
harfbuzz version: 7.1.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\blank\Documents
User Documents Path: C:\Users\blank\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/blank/Documents/OpenSCAD/libraries
User Config Path: C:\Users\blank\AppData\Local/OpenSCAD
Backup Path: C:/Users/blank/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  C:/Users/blank/Documents/OpenSCAD/libraries
  C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/WINDOWS/fonts
  C:/Users/blank/.local/share/fonts
  C:/usr/local/share/fonts
  C:/usr/share/fonts
  C:/Users/blank/.fonts
  C:/usr/X11/lib/X11/fonts
  C:/System/Library/Fonts
  C:/Library/Fonts
  C:/Users/blank/Library/Fonts

GLEW version: 2.1.0
OpenGL Version: 4.6.0 NVIDIA 531.41
GL Renderer: NVIDIA GeForce GTX 1050/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: WGL
PNG generator: lodepng
```

**Additional context**
[test.txt](https://github.com/openscad/openscad/files/11375993/test.txt)

pca006132, 2023-11-19:
works now in latest master

closed 2023-11-19 by kintel

nophead, 2024-02-04:
I get this warning with the latest snapshot 2024.01.26 (git 358af340c)

```
WARNING: [manifold] Minkowski failed with error, falling back to Nef operation: CGAL ERROR: precondition violation!
Expr: ! running
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Timer.h
Line: 83
```

Older versions with manifold work. E.g. 2023.12.02 (git dbc52d79c)

The rendered result looks OK though. I can't see anything wrong with it but I have stop on first warning set so my test fails.

pca006132, 2024-02-04:
@nophead can you try e7cf02b and the commit after it, i.e. test if this is due to manifold update.

nophead, 2024-02-04:
They are more than 30 days old so the artifacts have been deleted.

t-paul, 2024-02-04:
I have triggered a rebuild of the 2 commits:
Before: https://github.com/openscad/openscad/commit/e7cf02bfaa27939c633ad2847869b89b68f12fc2 - https://github.com/openscad/openscad/runs/21200701406
After: https://github.com/openscad/openscad/commit/d212319d48e541ac76b054f6ea205e3f7cca2f32 - https://github.com/openscad/openscad/runs/21200704501

nophead, 2024-02-04:
Thanks, they both give the same error message.

nophead, 2024-02-06:
It is a shame because the older versions that don't give this error on Minkowski have the convexity bug.