openscad/openscad #4628: Build error. std::memmove warning. Specified bound exceeds maximum object size.

craftmaster1231, 2023-05-05:
Please first consider that general questions about using OpenSCAD or how to write code for specific models are better asked on the mailing list. See https://openscad.org/community.html for more details.

**Environment**
Compiling on Fedora release 38 .
GNU Make 4.4
cmake version 3.26.3
c++ (GCC) 13.1.1 20230426 (Red Hat 13.1.1-1)

**Issue**
Build error on Building CXX object submodules/manifold/src/collider/CMakeFiles/collider.dir/src/collider.cpp.o

because of std::memmove warning.
Specified bound exceeds maximum object size.

So, here is what I did.

### Cloned project:
`git clone https://github.com/openscad/openscad.git`
`git submodule update --init --recursive`

### Got Dependencies:
` sudo ./scripts/uni-get-dependencies.sh`

### Checked dependencies:
`./scripts/check-dependencies.sh`
[image]

### Fixed `qscintilla2` with:
 `sudo dnf install qscintilla-qt6`
`sudo dnf install qscintilla-qt6-devel`
[image]

### Then ran:
`mkdir build && cd build && cmake .. -DEXPERIMENTAL=1`
to generate Makefile, but gon an error:
[image]

Fixed it with:
`sudo dnf install cairo-devel`

Then run cmake again.

Some screenshots:
[image]
[image]
[image]
[image]

## After all, I ran make:
`make`

Here are screenshots of error:
[image]
[image]
[image]

thehans, 2023-05-05:
Openscad is still built against Qt5, I don't believe it works with Qt6 yet.

But the error looks like it is deep within [manifold](https://github.com/elalish/manifold) code which has its own git repo, included as a git submodule, and may need to be reported upstream there.

I would recommend to first test a build with `EXPERIMENTAL=OFF`, to separate manifold build issues from the rest of openscad build.

craftmaster1231, 2023-05-05:
> Openscad is still built against Qt5, I don't believe it works with Qt6 yet.
>
> But the error looks like it is deep within [manifold](https://github.com/elalish/manifold) code which has its own git [...]

Tried with `EXPERIMENTAL=OFF`
Got an error:
[image]
Should i try to `#include <cstdint>`  and push that change on success?

craftmaster1231, 2023-05-05:
> Tried with `EXPERIMENTAL=OFF` Got an error: [image] Should i try to `#include <cstdint>` and push that change on success?

Why am I first person catching that simple type of error?

thehans, 2023-05-05:
> Why am I first person catching that simple type of error?

You aren't #4626

craftmaster1231, 2023-05-05:
> > Why am I first person catching that simple type of error?
> 
> You aren't #4626

Checked for "stdin" word only in Issues, not Pull Requests. Just missed.
Thanks!

craftmaster1231, 2023-05-05:
Everything compiled perfectly after `#include <cstdint>` to `src/glview/OpenGLContext.cc` and passing `-DEXPERIMENTAL=OFF
` to cmake.

closed 2023-05-05 by craftmaster1231