openscad/openscad #4631: When to use a console message, and when to use a dialog box

jordanbrown0, 2023-05-09:
In #4600 a user is confused because "export to STL" wasn't doing anything.  It turned out that the problem was that the user hadn't done a render, and so **on the console** OpenSCAD was displaying "ERROR: Nothing to export! Try rendering first (press F6)".  The user simply hadn't noticed this message on the console.

In my opinion, this is a UI design error.  Having **execution** warnings and errors appear in the console seems natural, but when you do a UI action - click a button, select a menu item, use a keystroke shortcut - if there is an error then it seems like the result should be a dialog box that either tells you about the problem and you then dismiss, or tells you about the problem and guides you through resolving it.

In the case in #4600, it seems to me that the minimum should have been a dialog box with that message in it, but better still would have been for that dialog box to have a RENDER button that would render the model and then export it.

gsohler, 2023-05-10:
Of Course I admit that it would be better to have this warning message not in the console, but having something which pops into your eyes.
But after all, OpenSCAD is a programming language: You insert your code, you watch your model and  in case it does not look expected you watch into the console more more details.
Personally I don't bear with the user which did not read the message but I agree that it would be better to make a programmer's tool more windows like ;)

jordanbrown0, 2023-05-10:
The distinction is between messages that result *from your program* and messages that result *from your actions in the UI*.  Messages that result from your program belong in the console; messages that result from UI actions belong in the UI, typically as popups.

MethylBromide, 2024-02-19:
I'm with Jordan on this.

MethylBromide, 2025-02-16:
And in addition: we lack a good way to respond to illegal inputs in the Customizer. This too is the result of a user action rather than a code issue, and the choices on how to respond in the script are:

- Log a warning using echo, which the user might not see.
- Log a warning using assert, which the user might be a little more likely to see but which unnecessarily adds a stack trace.
- Create some "ghost" text in the preview window, which the user is more likely to notice but which is a challenge to position so that it doesn't intersect any objects in the coordinate space.