openscad/openscad #4634: Polyhedron add points mark

sonichy, 2023-05-10:
```
polyhedron(
 points = [[0,-50,60], [0,50,60], [0,50,0], [0,-50,0], [60,-50,60], [60,50,60]],
 faces = [[0,3,2], [0,2,1], [4,0,5], [5,0,1], [5,2,4], [4,2,3], [4,3,0], [5,1,2]],
 showPoints = true //add
);
```
[image]

UBaer21, 2023-05-10:
there are libraries doing that 
[image]

For a beginner this might seem useful but consider a polyhedron with 10,000 or more points so you need more settings for orientation. Also it is useful to have normal vectors for faces.

gsohler, 2023-05-10:
Great library! Whats ITS Name ?
And how can this library Render 2d Text into 3d space for use with openscad?
Openscad uses artificial Text for  the Scale Matket annotation.

UBaer21, 2023-05-10:
@gsohler  this example is from [ub.scad](https://github.com/UBaer21/UB.scad/tree/main/examples/UBexamples)
But there are also others like [BOSL](https://github.com/revarbat/BOSL/wiki/debug.scad#debug_vertices)

2D text is always a 3D object in preview and can be moved and rotated - but you also can extrude the text to make it a real 3D object. ( the library automatically orient and scale the text according to the viewpoint and hides it when render F6)

sonichy, 2023-05-11:
Simple solution
```
points = [[0,-3,0], [-2,-3,0], [0,-3,3], [-2,0,0], [0,0,3], [0,0,0]];

for(i=[0:len(points)-1]){    
    translate(points[i]){
        rotate([90,0,0])
        linear_extrude(height=0.05) {            
            text(str(i), size=0.5);
        }
    }
}
    
polyhedron(
    points = points,
    faces = [[1,3,2], [2,3,4], [0,2,4], [4,5,0], [0,1,3], [0,3,5], [0,1,2], [3,4,5]]
);
```
[image]

closed 2023-05-11 by sonichy

jordanbrown0, 2023-05-11:
For extra cleverness, something based on ...
```
rotate($vpr) linear_extrude(height=1) text("hello");
```
will keep the text rotated to face the user.  (Alas, it does not track rotation; you have to F5.)  But note this trick won't work if it's buried inside other transformations.