openscad/openscad #4635: polyhedron render disappear

sonichy, 2023-05-10:
```
cube(10);

polyhedron(
 points = [[0,-3,0], [-2,-3,0], [0,-3,3], [-2,0,0], [0,0,3], [0,0,0]],
 faces = [[1,3,2], [2,3,4], [0,2,4], [4,5,0], [0,1,3], [0,3,5], [0,1,2], [3,4,5]]
);
```

UBaer21, 2023-05-10:
no 
[image]

UBaer21, 2023-05-10:
faces 4,5 and 7 have wrong order resulting in inverted normal
[image]

sonichy, 2023-05-10:
What is the order ?

UBaer21, 2023-05-10:
a face has two sides .. an inside (pink in F12 view)  and an outside (white) , this is the normal vector and depends if the points are cw or ccw ordered 
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#Mis-ordered_faces

jordanbrown0, 2023-05-10:
Faces need to be specified clockwise when viewed from the outside.  Use View/Thrown Together (F12); in the default color scheme anything purple is backwards.

[image]

closed 2023-05-10 by thehans

sonichy, 2023-05-11:
https://files.openscad.org/documentation/manual/Primitive_Solids.html#Mis-ordered_faces