openscad/openscad #4638: VsCode and openscad

AlexDep, 2023-05-14:
**Describe the bug**
Indicate a line where the error is but there is no code at this 

[image]

**To Reproduce**

Used VSCode with "Openscad Language support" extension from "Leathong" and the last Openscad version on a MacOs (last version.

Using following modules :
- arc-module-for-OpenSCAD/arc-module.scad
- and Round-Anything-OpenSCAD/polyround.scad

**Expected behavior**
Should tell where the error is.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

UBaer21, 2023-05-14:
likely some bracket is not closed so the code ended without and the error is after the last possible position.
[image]

kintel, 2023-05-14:
This extension isn't developed by us. Looks like this is the place to report it: https://github.com/Leathong/openscad-LSP

closed 2023-05-14 by kintel