openscad/openscad #4640: OSX brew cmake error - missing: HARFBUZZ_INCLUDE_DIRS

alidaf, 2023-05-17:
Since there is no up-to-date OSX build I'm trying to follow the instructions to compile on OSX using brew. The cmake fails with the above error.

This seems identical to issue 4328 but that was closed after the issuer managed to clear the error by deleting the build and source folders, and retrying. This doesn't work for me.

OSX Ventura 13.3.1  Intel

thehans, 2023-05-17:
The most up to date builds are under "Development Snapshots" of the main website:
http://openscad.org/downloads.html#snapshots
Latest OSX build is currently: https://files.openscad.org/snapshots/OpenSCAD-2023.04.17.dmg

alidaf, 2023-05-18:
That's a bit behind the Linux and Windows versions, which is why I tried compiling. I'm chasing any improvements to Manifold.

kintel, 2023-05-18:
Could you clarify how you're trying to building OpenSCAD, including your Xcode version? You're writing "on brew"; are you trying to build brew packages?

kintel, 2023-05-22:
@alidaf macOS snapshots work again: https://openscad.org/downloads.html#snapshots

alidaf, 2023-05-22:
Sorry, I don't seem to be getting notifications via email. I'm basically following the instructions for building. I've tried both methods - homebrew and compiling directly. I've had issues with both, but in any case, snapshots are working again so Woot!